openscad/openscad #1127: Root Modifier (!) evaluates other code

MichaelAtOz, 2015-01-01:
Not sure if this is expected.
```
module a()
  echo("a");

module b()
  echo("b");

!a();
b();
```

Produces
```
Compiling design (CSG Tree generation)...
ECHO: "a"
ECHO: "b"
Compiling design (CSG Products generation)...
```

I would not expect the ECHO: "b".
Could be bad if a module was doing a lot of echoing, eg BOMs, and you were just interested in other modules.
With 2014.03 & 2014.12.30.

kintel, 2015-01-02:
The behavior of the ! operator is to evaluate everything, but only _render_ the tagged object. I guess we could change that. I feel there are some potential cans of worms involved in doing this properly, so it needs to be done carefully.

MichaelAtOz, 2015-01-17:
On reflection, you would have to evaluate everything in case variables are used in the tagged code.

kintel, 2015-01-17:
@MichaelAtOz The question is more if echo() should output stuff during compile time or render time.

thehans, 2019-03-05:
I also noticed this recently, and think it would be nice if we can avoid instantiating nodes outside of root modifier, so something like this could work:
```
module recur() recur();
recur();

!cube();
```
Infinite recursion is just a quick example, but you could imagine the call to recur as any other possibly expensive-to-instantiate chunk of deeply nested nodes, etc.

I think the best solution would be to determine the root node from within the parser.  Also doing the warnings for multiple root modifiers inside the parser should fix issues of having ! inside a loop etc.  (related: #1815)  and would avoid having an extra pass through nodes that `AbstractNode::find_root_tag` does.

t-paul, 2019-11-22:
Closing, `echo()` needs to be in evaluation phase or it would not be usable as debugging aid for recursive functions.

closed 2019-11-22 by t-paul