openscad/openscad #1128: Up-sized numbers

MichaelAtOz, 2015-01-01:
The manual describes numbers as;

>OpenSCAD has only a single kind of number, which is an IEEE floating point number. [OpenSCAD does not distinguish integers and floating point numbers as two different types, nor does it support complex numbers.] Because OpenSCAD uses the IEEE floating point standard, there are a few deviations from the behaviour of numbers in mathematics:
>
The largest representable number is about 1e308. If a numeric result is too large, then the result can be infinity (printed as inf by echo).
The smallest representable number is about -1e308. If a numeric result is too small, then the result can be -infinity (printed as -inf by echo).
If a numeric result is invalid, then the result can be Not A Number (printed as nan by echo).
If a non-zero numeric result is too close to zero to be representable, then the result will be -0 if the result is negative, otherwise it will be 0. Zero (0) and negative zero (-0) are treated as two distinct numbers by some of the math operations, and are printed differently by 'echo', although they compare equal.
>
Note that 'inf' and 'nan' are not supported as numeric constants by OpenSCAD, even though you can compute numbers that are printed this way by 'echo'. You can define variables with these values by using:
>
inf = 1e200 * 1e200;
nan = 0 / 0;
echo(inf,nan);"

I was fiddling with this code;
```
module flatten() {
inf = 1e4;
difference() {
	children();
	translate([-inf,-inf,-inf]) 
		cube([2*inf,2*inf,inf]);
        }
}

flatten() 
{
	cube(30,true);
	cylinder(d=10,h=100,center=true);
}
```
and basically wanted a huge cube, so theoretically it could cut any sized children.

I initially tried inf=1e200*1e200, it didn't produce a cube, then I tried just 1e200, then started low and got cubes with 1e4, but not 1e5.

Testing just with big cubes produced cubes for larger numbers, so
Issue A. difference() is not handling cubes bigger than ~1e4.

While fiddling I came across other issues;

Issue B. F6 Assertion violation with big cube (or difference) ie >1e13
```
Compiling design (CSG Tree generation)...
ECHO: inf = 1e+050, 2e+050
ECHO: inf = 100000, 200000 <== other bug, ignore
Rendering Polygon Mesh using CGAL...
CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: pe != 0
File: /data/OpenSCAD/libraries-mingw64-master/mxe-w64/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h
Line: 212
Geometries in cache: 32
Geometry cache size in bytes: 26016
CGAL Polyhedrons in cache: 9
CGAL cache size in bytes: 694656
Total rendering time: 0 hours, 0 minutes, 0 seconds
   Top level object is a 3D object:
   Simple:        yes
   Vertices:       72
   Halfedges:     216
   Edges:         108
   Halffacets:     80
   Facets:         40
   Volumes:         2
Rendering finished.
```

Issue C. 
I also found issues with plain cubes, in Orthogonal mode 4/6 views (the back/front/left/right buttons) show as blank with size 1e6 (oops double all these numbers ie 2e6 - the code was inf*2 see below), the cube is there just not showing for both F5 & F6. With size 1e15 Perspective loses the same views, at 1e16 top view goes (for F5 not F6), [ strangely,  Orthogonal bottom view returned ~1e16]. At 1e17 & Perspective all views are gone.

Similar issues with cylinder (1e33), sphere (goes at 1e6) square (1e36), circle (1e18).

The manual says the largest 'representable number', is there a largest Geometry number? (but 1e6 is not that big).

```
module f() {

	inf = 1e16;
	echo(inf=inf,inf*2);
	translate([-inf,-inf,-inf]) 
		cube([2*inf,2*inf,inf]);
}

!f();
```

laird, 2015-01-01:
As a mathematician, the expression "2*inf" was pretty amusing, since 2*infinity=infinity, though of course numerically (since inf isn't infinity but a large number) it makes sense. :-)

I started out doing the 'flat' with a large cube (though not infinity) and it works, technically, but it causes Customizer's viewer to get confused, since it includes the subtracted cube in the volume shown. And since fairly often I write for Customizer, I have gotten in the habit of trying to make subtracted shapes as small as possible. Your code is a good reminder that this isn't an OpenSCAD requirement, but a Customizer requirement, and ideally Customizer should be fixed so it only cares about the final generated shape, not the subtracted shapes.

kintel, 2015-01-02:
IssueA: This is a Preview problem. You're trying to render very large objects with OpenGL, which operates on a 32-bit depth buffer. We clip objects to avoid jeopardizing the resolution of the depth buffer. If you zoom far enough out, you'll see that the difference works, even with ```inf=1e6```. We could probably improve the renderer.

Issue B: I cannot reproduce that. With which version of CGAL is this?

Issue C: This is the same as A; disappearing objects due to clipping. A fix for this would be do adjust clipping dynamically. That won't fix A though, where we need to render object spanning the entire Z buffer range (since we want to zoom to view the smaller object).

kintel, 2015-01-02:
Suggestion: Split out the F6 issue as it's a different cause than the others,

MichaelAtOz, 2015-01-02:
> Issue B: I cannot reproduce that. With which version of CGAL is this?

OpenSCAD Version: 2014.12.30
System information: Windows 7 (v6.1) 2 CPUs 3.50 GB RAM
Compiler, build date: GCC "4.9.2" 64bit, Dec 30 2014
Boost version: 1_57
Eigen version: 3.2.2
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 5.4.0
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.42.1

use inf=1e14 and flush caches then F6, happens every time 4 me.

> (since we want to zoom to view the smaller object).

Note C was plain cube() ie;

```
module f() {

	inf = 1e13;
	echo(inf=inf,inf*2);
	translate([-inf,-inf,-inf]) 
		cube([2*inf,2*inf,inf]);
}

!f();
```
and cylinder() etc.

MichaelAtOz, 2017-05-16:
I can no longer reproduce the B. CGAL error in a late snapshot, assuming it's fixed.