openscad/openscad #1130: Error when projecting rotated object

meshulam, 2015-01-02:
I get an error trying to project the XZ face of an imported .stl model:

```
projection(cut=true) 
  rotate(a=90, v=[1, 0, 0]) 
    import(MODEL_FILE, convexity=10);
```

This results in no slice being rendered, and the following error in the console:

```
Module cache size: 7 modules
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometry Cache insert: import(file="../Coffee table/attempt3-li (14208 bytes)
Geometry Cache hit: import(file="../Coffee table/attempt3-li (14208 bytes)
Geometry Cache insert: multmatrix([[1,0,0,0],[0,6.12323e-17,-1, (14208 bytes)
ERROR: CGAL error in CGALUtils::project during bigbox intersection: CGAL ERROR: assertion violation! Expr: cet->get_index() == ce->twin()->get_index() File: /Users/matt/code/openscad/../libraries/install/include/CGAL/Nef_3/SNC_external_structure.h Line: 1169
WARNING: projection() failed.
Geometry Cache insert: projection(cut=true,convexity=0){multmat (0 bytes)
Geometry Cache hit: import(file="../Coffee table/attempt3-li (14208 bytes)
Geometries in cache: 3
```

I couldn't really figure out what the offending code is doing, but the very small (6e-17) value printed in the log above makes me think maybe a floating point bug in the cache lookup? Just a wild guess.

kintel, 2015-01-02:
This sounds model dependent. Could you include your STL file?

meshulam, 2015-01-02:
It was exported from Blender: https://www.dropbox.com/s/qqp646kn9p8cxht/attempt3-lines.stl?dl=0

Also, it seems to work fine with rotation angles arbitrarily close to +/- 90 degrees, just not exactly 90. a=89.999 projects correctly.

kintel, 2015-01-02:
Confirmed. It fails somewhere inside CGAL's intersection code. Apparently that functions isn't as robust as we'd hoped for. I'll try to slim down the STL to a minimal example triggering the issue and look into it.

meshulam, 2015-01-02:
I can try making simplifications to the STL in Blender too.

meshulam, 2015-01-03:
Turns out it was a messed up geometry, several faces were overlapping and the back of one of them was exposed. Here's an isolated example that still trips the error: https://www.dropbox.com/s/uipfibe2ss8fht4/projection-error.stl?dl=0

kintel, 2015-01-20:
See http://cgal-discuss.949826.n4.nabble.com/Assertion-failed-intersect-Nef-polyhedron-with-plane-tt4660337.html

thehans, 2020-06-04:
This bug is pretty old and unfortunately the dropbox links have expired so there's no way to tell if its still reproducible in a recent release. 
@meshulam Any chance you still have these files or remember how to recreate?

gist.github.com accepts stl files by the way, and i don't believe they will expire.

meshulam, 2020-07-18:
@thehans Unfortunately I couldn't find the STL files that I used to reproduce this issue :-(

thehans, 2020-07-18:
Thank you for the reply.
It is unfortunate, but In that case I'm closing this as not reproducible.

closed 2020-07-18 by thehans