openscad/openscad #1141: Difference disappearing Mac OSX 10.4

alana000, 2015-01-06:
I made 2 shapes using openscad and differenced one from the other. My issue is when I compiled F5, one part had disappeared, however when I rendered F6 i've got the shape I want. 
I'll explain with photos :)
[image]
I want to difference the above image from this:
[image]
this is how it renders F5
[image]
but this is how it compiles :( 
[image]

The end problem is that I want to upload it to thingiverse customiser but the customiser software is only reading it as the compiled object. It exports from openscad to an .stl just fine. 

I had a read of the thread here https://github.com/openscad/openscad/issues/198, where they seem to have the same problem and tried forcing gold feather rendering but it didn't solve the issue.

Just a disclaimer I'm a totally newbie to openscad so it's possible i'm doing something very obvious wrong. 
Any help would be majority appreciated and thanks in advance :)

kintel, 2015-01-06:
Just to clarify: Are you really using Mac OS X 10.4? Which version of OpenSCAD are you using?

alana000, 2015-01-06:
ah lol sorry, missed the .9 
it openscad version 2014.03

kintel, 2015-01-06:
Good :) Could you share your .scad file?

alana000, 2015-01-06:
sure thing:

difference() {
cube(size = [45.51,45.51,12.25], center = true);translate([22.755,0,-9.125]) 
scale([0.16,0.45,0.55]) sphere(r=23.86);translate([-22.755,0,-9.125]) 
scale([0.16,0.45,0.55]) sphere(r=23.86);
}

difference() {
union(){
translate([0,0,-27.415]) cube(size = [35.96,35.96,47.18], center = true);
translate([0,17.48,-36.125])cylinder(h=30, r=1.375, $fn=30);
translate([0,-17.48,-36.125])cylinder(h=30, r=1.375, $fn=30);
translate([17.48,0,-36.125])cylinder(h=30, r=1.375, $fn=30);
translate([-17.48,0,-36.125])cylinder(h=30, r=1.375, $fn=30);
}

translate([0,0,-32.3])
group(){
difference() {
translate([0,0,-10]) cube(size = [40,40,20], center = true);
polyhedron(
  points=[ [18.915,18.915,],[18.915,-18.915,0],[-18.915,-18.915,0],[-18.915,18.915,0],[0,0,-18.92]  ],faces=[ [0,1,4],[1,2,4],[2,3,4],[3,0,4],[1,0,3],[2,1,3] ]
 );
}
}
}

kintel, 2015-01-06:
I noticed that your polyhedron has it's indices reversed. You can see this in Throwntogether mode (F12) - inverse faces are rendered pink. Try replacing it with this:
```
polyhedron(
points=[ [18.915,18.915,],[18.915,-18.915,0],[-18.915,-18.915,0],[-18.915,18.915,0],[0,0,-18.92] ],faces=[ [4,1,0],[4,2,1],[4,3,2],[4,0,3],[3,0,1],[3,1,2] ]
);
```

Not sure if that could trigger your problems though.
Btw., what hardware are you using? You can paste Help->Library Info here.

alana000, 2015-01-06:
woah, it worked! thank you so much! 

just fyi library info is:
OpenSCAD Version: 2014.03
Compiler, build date: Clang "5.1 (clang-503.0.40)", Apr 21 2014
Boost version: 1_54
Eigen version: 3.2.0
CGAL version, kernels: 4.3, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 4.8.5
MingW build: No
GLib version: 2.38.2
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Users/Abirch/Documents/OpenSCAD/libraries
  /Applications/OpenSCAD.app/Contents/MacOS/../Resources/libraries
GLEW version: 1.10.0
OpenGL Version: 2.1 ATI-1.24.35
GL Renderer: AMD Radeon HD 6770M OpenGL Engine
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix

thanks again :D

closed 2015-01-06 by kintel

alana000, 2015-01-06:
ahh just one more question; when I open the scad file with thingiverse customizer it's still missing the bottom part.(see below) Any idea why that might be?
[image]

MichaelAtOz, 2015-01-06:
> I noticed that your polyhedron has it's indices reversed. You can see this in Throwntogether mode (F12) - inverse faces are rendered pink.

@kintel perhaps it is time to show the pink in preview, seems the new user base is increasing. ??
(BTW do you have any metrics on the user base?)

MichaelAtOz, 2015-01-06:
BTW
```
translate([0,0,-32.3])
group(){
difference() {
```
group() is not really part of the language AFAIK, but it seems to do something, should be union(), but...that doesn't fix the CustomiZer issue.

At first look I can't see what the issue is. If nobody with a snappier mind can, then there is the Thingiverse [Google group](https://groups.google.com/forum/#!forum/thingiverse), tho they have been a little less responsive lately...

alana000, 2015-01-06:
thanks for having a look, I'll give the google group a try :)

tbuser, 2015-01-06:
Sorry, Customizer is still running openscad 2013.06.  If you replace the word faces with triangles in the polyhedron call, it should work. (in both)  See: http://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#polyhedron

kintel, 2015-01-06:
@tbuser Any plans for upgrading? You're almost two versions behind.

tbuser, 2015-01-07:
I know.  :(  Soon, I hope.