openscad/openscad #1145: Clarify Export as STL icon

kintel, 2015-01-08:
The "Export as STL" icon should make it more clear that we're exporting STL, perhaps replacing the arrow with the text "STL" ?

TDeagan, 2015-01-12:
* Is this the icon in the menu or are you contemplating icons on the toolbar for this?
* Are icons needed for the other export options (OFF, AMF, etc.)?  Could they share the motif of the extension replacing the arrow?
* Is this in progress by anyone or is it useful if I jump on this?

t-paul, 2015-01-12:
There's currently a single export icon in the toolbar which does STL export only. For the release we probably only need the STL icon.

I tried to create one, but could not come up with something that looked nice in 32x32.

For later I guess it depends on how #649 evolves. Probably we will need to have an export dialog that can select the file format and also modify format specific parameters. It still might make sense to leave the different menu entries there and initialize the dialog accordingly.

TDeagan, 2015-01-12:
Urp, sorry, I'd had the editor hidden so often that I missed the toolbar above it!  I was looking at the viewport toolbar only.

I'm fairly sure Qt offers the ability to  drop down a set of icons from a single icon if you wanted the existing icon to be a generic export and the icons below to represent the different formats.  Later it could serve as the trigger for the dialog.

For me, I'd use the STL export 50:1 over the other formats (until multi-material AMF is ready :) so I can understand the value of the STL export as a single icon.

I'll take a shot at a 32x32 and post it so you and the krewe can critique.

TDeagan, 2015-01-12:
Maybe something like:
[image]

kintel, 2015-01-12:
Looks good to me!

As discussed earlier, I agree with Tim that it would be cool to have some sort of drop down icon for this, and e.g. let the last used one be the default one. The same technique could be used to make the axis view buttons take less space. I'm not sure how much work it is to convince Qt to be happy about this though.

t-paul, 2015-01-12:
It's a bit hard to read on retina display, but I guess that's more a problem of me using tiny icons than the icon itself ;).
The icon will need a transparent background and we need a white one too (Easy to do in GIMP based on the black one using Colors->Levels and swapping the output levels).

If there are no objections, I can do that and add the icons to the gui-updates branch where the other icon changes are already collected...

TDeagan, 2015-01-12:
Sounds good to me.  I did the icon in an icon editor tool, I have a transparent version as well, I can post it when I get home from work unless you've already stripped the background off the one above.  GIMP will always do the job.

t-paul, 2015-01-12:
Pushed to gui-updates (#1150) https://github.com/openscad/openscad/commit/47254dd021bc86d6c6fead5a0f0ba54814d8e933

closed 2015-01-16 by kintel