openscad/openscad #1146: Error using Surface command

magotheobroma, 2015-01-08:
When I use the Surface command always I get an error.
As example, with this code
[image]
I get this error:
[image]

MichaelAtOz, 2015-01-08:
surface() with .png is only supported in development snapshots ATM. Looks like you are on 2014.03?

magotheobroma, 2015-01-08:
Yes, I'm on 2014.03. Thanks.

Data: Thu, 08 Jan 2015 01:25:19 -0800
Oggetto: Re: [openscad] Error using Surface command (#1146)

t-paul, 2015-01-08:
The crash should be fixed too, trying with some random data (using /dev/urandom) a warning message is shown as expected.
```
WARNING: Illegal value in '/dev/urandom': bad lexical cast: source type value could not be interpreted as target
```

closed 2015-01-08 by t-paul