openscad/openscad #1149: GUI issues

t-paul, 2015-01-09:
- [ ] Some icons are not working with a dark theme
- [ ] The font for the info window should only use bold so it reacts to user settings

[image]

MichaelAtOz, 2015-01-09:
Should the extra warning ONLY show when the console is hidden?

t-paul, 2015-01-09:
No, I don't think it should depend on the console visibility. There might be other ways like hide it after some time or even make it configurable.

t-paul, 2015-01-09:
See #1150...

closed 2015-01-09 by t-paul

MichaelAtOz, 2015-01-09:
> There might be other ways like hide it after some time or even make it configurable.

Perhaps when there is activity in the editor window?

But configurable would be good, I usually have the console open so it would be superfluous for me.