openscad/openscad #1155: Feature Request: Export to SCAD polyhedron?

lordofhyphens, 2015-01-13:
I'very been wondering about being to export a compiled design to a set of polyhedron statements (one per separate object), with points and faces. I believe that it will help facilitate modifying existing STLs and provide a pseudo-debugging output. Additionally, I believe it would permit a more iterative work flow.

NateTG, 2015-01-13:
The STL files are already really close to that, and it's pretty easy to write an external utility that does that.

Can you give an example of what you want to do with a feature like that?

lordofhyphens, 2015-01-13:
I'd like to investigate iterative compilation--using multiple passes to build objects on top of objects (similar to how multiple LaTeX passes are used for cross-references in a manuscript). Currently, the way to put something "on top" of some other complex object is to step back and do a stack of math or guess/check repeatedly. An export to a polyhedron() statement would let us precompute the geometry and import it into another design that builds on it.

Doing everything with SCAD primitives that are readable, I think, will give be better reliability and transparency than import(). I also find myself wanting to do some manipulation of STLs I don't have the source for, and polyhedron() is a lot more readable than the raw STL format, especially if that STL is binary and not text.

t-paul, 2015-01-13:
I guess the geometry feedback part is discussed in #301 already.

For hand editing, AMF might be similar complex as a polyhedron (or maybe even easier as it's possible to process it with command line XML tools).

lordofhyphens, 2015-01-13:
I wasn't aware that import() allowed AMF files.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

t-paul, 2015-01-13:
The AMF import is waiting to be merged (probably after the release is out).

jmerc77, 2023-08-29:
Similar idea, but directly access points and faces from a module instead of needing to export then import. Wasn't sure if anyone knew they wanted this, though. Also thought it would be cool if it worked for 2d modules, too.

gsohler, 2023-08-29:
Such a Feature exists in my OpenSCAD Fork. The function is called 'mesh'.
Its only available on Python Side yet.it Returns a tuple with Point Array
and vertices, similar to polyhedron Input.
'mesh' can be Used together with polyhedron to create fancy translations.

t-paul, 2023-08-29:
I believe this is really just asking for an export format that produces a single file containing a single `polyhedron()` call that is a valid scad script.