openscad/openscad #1161: concat and assign doestn work properly

mirecta, 2015-01-15:
i try write function bezier which fill points of bezier into array and return it

my code

```
 module bezier(array, P1,P2,P3,P4, step=100){
    
    result = array;
    
    for ( i = [0 : 1/step : 1] )
{
    temp = [[
    (1-i)*(1-i)*(1-i)*P1[0] + 3*i*(1-i)*(1-i)*P2[0] + 3*i*i*(1-i)*P3[0] + i*i*i*P4[0],
    (1-i)*(1-i)*(1-i)*P1[1] + 3*i*(1-i)*(1-i)*P2[1] + 3*i*i*(1-i)*P3[1] + i*i*i*P4[1]
    ]];
    result = concat(result,temp);
    echo(result);
}
echo(result);
//return result;
 
}
bezier([],[0,0],[1,1],[2,1],[3,0]);
```

last echo print empty vector 
and echo inside loop print only actual concat temp with array

mirecta, 2015-01-15:
the root of evil is 
```
a = 0;
for ( i = [0:1:10]){
    a = a+1;
echo(a);
}
```
without this many things is impossible

nophead, 2015-01-15:
No, everything you could do with iteration and mutable variables can be done with recursion without mutable variables.

mirecta, 2015-01-15:
but what is more simple to use , to memory etc ...  recursion or iteration ?

MichaelAtOz, 2015-01-15:
We really need to improve the wiki re this, explain the rational better.

It is not an iterative program it is a declarative tree structure...

I've had some ideas, but they need diagrams of tree structure, I'll try to have a go, but don't hold your breath...

t-paul, 2015-01-15:
See http://en.wikibooks.org/wiki/OpenSCAD_User_Manual/List_Comprehensions
```
function bezier(P1,P2,P3,P4,step=100) = [
    for ( a = [0 : step] ) let (i = a / step) [
        (1-i)*(1-i)*(1-i)*P1[0] + 3*i*(1-i)*(1-i)*P2[0] + 3*i*i*(1-i)*P3[0] + i*i*i*P4[0],
        (1-i)*(1-i)*(1-i)*P1[1] + 3*i*(1-i)*(1-i)*P2[1] + 3*i*i*(1-i)*P3[1] + i*i*i*P4[1]
    ]
];

polygon(20 * bezier([0,0],[1,1],[2,1],[3,0]));
```

Also someone wrote a spline library, currently sitting as pull request there: openscad/scad-utils#1

mirecta, 2015-01-15:
my point was make better 2d subsystem 
so at start i have empty vector of points and made some functions to draw 2d elements for example
```
points = [];

points = bezier(points,P1,P2,P3,P4);
points = arc(points,P1,P2,R);
points = line(points,P1,P2);
.......
......
.......
polygon(points);

```
u can partialy draw complex shapes caling simple methods

t-paul, 2015-01-15:
I agree, the 2D system could be extended. Issues for that topic already exist, e.g. #508 and #956. Having non-closed 2D primitives to build up closed polygons could make the ```text()``` and ```trace()``` prototype (#1110) more general.

The functionality you've sketched in the last comment is already possible, just the reassignment to the same "variable" will not work, see https://github.com/openscad/list-comprehension-demos for some examples.

closed 2015-01-15 by t-paul