openscad/openscad #1163: Render crash on lego block

trayracing, 2015-01-16:
Crash when render a simple lego brick:
OpenSCAD 2014.12.30 (git 8d4d83f) on windows 8.1

1)download this: http://www.thingiverse.com/thing:5699/#files
2)open file uploads-46-fe-67-38-94-block-120713.scad
3) render (f6) - this should work
4) change call to "blocks" on line 16, 2nd parameter (width) from 5 to 15
5)render. This crashes (preview works fine)

process peaks at 142 MB of memory.

kintel, 2015-01-16:
I cannot reproduce that here. Could you try with the most recent development snapshot?
(There's been a bunch of changes since then - release sprint).

trayracing, 2015-01-16:
Just retried with version 2015.01.13 (git 3666636).
Still crashes 32 bit & 64 bit. 
I did check the OpenSCAD advanced rendering options and the are the defaults.

However, I tried second machine and it does not crash.
Primary difference I see is the crasher CPU is an AMD A8-4500M and the non-crasher is an Intel i7-4510U. Both use integrated graphics and have the same memory size and OS.

kintel, 2015-01-16:
Oh, CPU-specific crash - that's fun..
Just as a datapoint: Just tested on a 32-bit Win 8.1 VM (running on Intel) and it didn't crash there either.

Now, do you have access to any sort of info about the crash?
Does it crash if you use render the crashing design without first rendering the working design?

MichaelAtOz, 2015-01-16:
Does NOT crash on W7/64 Intel Core-duo NVIDIA ION graphics, OpenSCAD 2015.01.13 64bit.

trayracing, 2015-01-16:
Yes, it did crash in the working design.

But - close this puppy!
I managed to snag a minidump and second stack frame from the top was in ati06axx.dll, which is an AMD dll. Although my machine was claiming to be up-to-date, installing the lastest release from AMD solved it. (Don't know if it was on the CPU or GPU side of the APU.)

closed 2015-01-16 by trayracing