openscad/openscad #1164: Create Full Set of Export Icons

TDeagan, 2015-01-16:
I wanted to keep the thread moving regarding the creation of icons for all export formats beyond just STL.

I've worked with an icon editor to try and create a more readable icon at the tiny menu scale. Here's a candidate:
[image]

Here's what the STL icon looks at slightly larger size on the toolbar:
[image]

I've also been experimenting (semi-successfully) with creating a drop down icon menu from the toolbar Export STL (to become Export) Icon.  That's no where near ready and could be a different 

Sorry to keep whacking at the STL export icon after just getting a completed merge with it, but when I tried the same pattern with the other export format extensions (AMF, OFF, etc.) they just didn't fit/work.  They were unreadable.  I had to find a different layout with a bigger font.

If this is appealing I can:
A. Create a pull request with the icons and their inclusion in the menu
B. Submit just the icon images for someone else to do with as they will
C. Hold off until I can pitch some kind of drop down icon for exports on the toolbar and do these two changes together.
D. Put the entire issue to bed for a while and work on something else.

Suggestions appreciated.

Also, I've got white and black versions but am not stumbling across where the white ones are used.

kintel, 2015-01-16:
Nice.
I do have one (annoying) suggestion: The icons look similar to the "New" icon and it isn't obvious to me that it's about exporting unless I read the tooltip. The disk icon might be a better starting point, but may not work well mixed with text.
Nevertheless, it's still an improvement, so I'd merge it :)

t-paul, 2015-01-16:
The white icons are [automatically](https://github.com/openscad/openscad/blob/65e50b2be63587fcd1d38ca9e7e6db313c6bb1e8/src/mainwin.cc#L477) used when a dark theme is detected. (See screenshot in #1150 from my kubuntu test VM).

I also like the new icons even better. I see nothing wrong with improving after something is merged.

TDeagan, 2015-01-16:
First attempt at a useful pull request at #1166

closed 2015-01-20 by kintel