openscad/openscad #1169: Bad STL output for Prusa's

lordofhyphens, 2015-01-20:
Gettting degenerate (and otherwise weird) export results when trying to build Prusa's i3 repository, particularly x-end.stl. Slic3r spots a host of manifold errors as well.

Picture of glitch:
[image]

picture of AMF:
[image]

I think this is a regression, I don't remember having this issue with 2014.03 (or thereabouts), but the problem is there as of 2014.09 and current git HEAD.

I don't see this behavior when using amf export instead (same makefile, just changed the STL file extension to AMF). 

snippet from build log:
Geometry Cache insert: cube(size=[36,5,14],center=true); (672 bytes)
CGAL Cache insert: group(){group(){group(){multmatrix([[1,0 (117616 bytes)
CGAL Cache insert: difference(){cube(size=[36,5,14],center= (69184 bytes)
CGAL Cache insert: group(){difference(){cube(size=[36,5,14] (69184 bytes)
CGAL Cache insert: group(){group(){difference(){cube(size=[ (69184 bytes)
CGAL Cache insert: multmatrix([[1,0,0,18],[0,1,0,2.5],[0,0, (69184 bytes)
CGAL Cache insert: group(){multmatrix([[1,0,0,18],[0,1,0,2. (69184 bytes)
CGAL Cache insert: group(){group(){multmatrix([[1,0,0,18],[ (69184 bytes)
CGAL Cache insert: group(){group(){group(){multmatrix([[1,0 (69184 bytes)
CGAL Cache insert: multmatrix([[1,0,0,-6],[0,1,0,0],[0,0,1, (69184 bytes)
Geometry Cache hit: multmatrix([[1,0,0,26],[0,1,0,9],[0,0,1, (3408 bytes)
CGAL Cache insert: group(){group(){group(){multmatrix([[1,0 (61264 bytes)
CGAL Cache insert: group(){group(){group(){multmatrix([[1,0 (37504 bytes)
CGAL Cache insert: multmatrix([[1,0,0,5],[0,1,0,5],[0,0,1,0 (29464 bytes)
CGAL Cache insert: group(){multmatrix([[1,0,0,-6],[0,1,0,0] (69184 bytes)
Geometry Cache hit: multmatrix([[1,0,0,26],[0,1,0,9],[0,0,1, (2112 bytes)
Geometry Cache hit: multmatrix([[1,0,0,0],[0,1,0,5],[0,0,1,- (1248 bytes)
CGAL Cache hit: group(){multmatrix([[1,0,0,16],[0,1,0,35 (382336 bytes)
CGAL Cache insert: union(){group(){group(){group(){multmatr (265344 bytes)
CGAL Cache insert: difference(){union(){group(){group(){gro (602488 bytes)
CGAL Cache insert: union(){difference(){union(){group(){gro (602488 bytes)
CGAL Cache insert: difference(){union(){difference(){union( (602488 bytes)
CGAL Cache insert: group(){difference(){union(){difference( (602488 bytes)
CGAL Cache insert: multmatrix([[2.22045e-16,-1,0,0],[1,2.22 (602488 bytes)
CGAL Cache hit: group(){difference(){union(){difference( (602488 bytes)
CGAL Cache insert: multmatrix([[1,0,0,0],[0,-1,0,0],[0,0,1, (602488 bytes)
CGAL Cache insert: multmatrix([[2.22045e-16,1,0,0],[-1,2.22 (602488 bytes)
CGAL Cache insert: multmatrix([[1,0,0,10],[0,1,0,-50],[0,0, (1017200 bytes)
CGAL Cache insert: multmatrix([[1,0,0,0],[0,1,0,50],[0,0,1, (1017200 bytes)
CGAL Cache insert: multmatrix([[1,0,0,67],[0,1,0,14],[0,0,1 (602488 bytes)
CGAL Cache insert: multmatrix([[1,0,0,77],[0,1,0,-14],[0,0, (602488 bytes)
CGAL Cache insert: group(){multmatrix([[1,0,0,10],[0,1,0,-5 (3239112 bytes)
WARNING: PolySet has degenerate polygons

lordofhyphens, 2015-01-21:
library info:
OpenSCAD Version: 2015.01.20
Compiler, build date: GCC "4.9.1", Jan 20 2015
Boost version: 1_55
Eigen version: 3.2.2
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 4.8.6
QScintilla version: 2.8.4
MingW build: No
GLib version: 2.42.1
Application Path: /home/grad/lenox/openscad
Documents Path: /home/grad/lenox/.local/share
Resource Path: /home/grad/lenox/openscad
User Library Path: /home/grad/lenox/.local/share/OpenSCAD/libraries
User Config Path: /home/grad/lenox/.config/OpenSCAD
Backup Path: /home/grad/lenox/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/grad/lenox/.local/share/OpenSCAD/libraries
/home/grad/lenox/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.10.0
OpenGL Version: 4.4.0 NVIDIA 340.65
GL Renderer: GeForce GTX 460/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_AMD_multi_draw_indirect
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
[... 232 lines cut ...]
GL_NV_texture_env_combine4
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_vdpau_interop
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum

lordofhyphens, 2015-01-21:
./check-dependencies from the other workstation I run OpenSCAD from and run into this problem

depname     minimum     found       OKness      
qt          4.4         5.3.2       OK          
qscintilla2 2.7         unknown     NotOK       
cgal        3.6         4.5         OK          
gmp         5.0         6.0.0       OK          
mpfr        3.0         3.1.2       OK          
boost       1.35        1.55        OK          
opencsg     1.3.2       1.3.2       OK          
glew        1.5.4       1.7.0       OK          
eigen       3.0         3.2.2       OK          
glib2       2.0         2.42.1      OK          
fontconfig  2.10        2.11.       OK          
freetype2   2.4         17.1.11     OK          
harfbuzz    0.9.19      0.9.35      OK          
gcc         4.2         4.9.1       OK          
bison       2.4         3.0.2       OK          
flex        2.5.35      2.5.39      OK          
make        3           4.0         OK

lordofhyphens, 2015-01-21:
another system with the same behavior, running the same distro (roughly, this is Debian Testing)
qt          4.4         5.3.2       OK          
qscintilla2 2.7         unknown     NotOK       
cgal        3.6         4.5         OK          
gmp         5.0         6.0.0       OK          
mpfr        3.0         3.1.2       OK          
boost       1.35        1.55        OK          
opencsg     1.3.2       1.3.2       OK          
glew        1.5.4       1.7.0       OK          
eigen       3.0         3.2.2       OK          
glib2       2.0         2.42.1      OK          
fontconfig  2.10        2.11.       OK          
freetype2   2.4         17.1.11     OK          
harfbuzz    0.9.19      0.9.35      OK          
gcc         4.2         4.9.1       OK          
bison       2.4         3.0.2       OK          
flex        2.5.35      2.5.39      OK          
make        3           4.0         OK

kintel, 2015-01-21:
I've tested on Mac OS X (various OpenSCAD versions) and Ubuntu 14.04 (current master), but I haven't managed to reproduce this.

kintel, 2015-01-21:
A shot in the dark: Could you try with the current master? Just in case your issue was related to #1165

kintel, 2015-01-22:
Any progress on this? If I could find a way to reproduce I could dig further into it..

lordofhyphens, 2015-01-23:
Still running through the bisect. Unfortunately, there's a lot of commits that I have to skip because there's a different bug that is causing a segfault or OpenSCAD itself fails to compile.

lordofhyphens, 2015-02-18:
Well, looks like the bug is fixed now! Excellent.

closed 2015-02-18 by lordofhyphens