openscad/openscad #1172: Syntax highlighting: # affects the whole line

kintel, 2015-01-22:
This is likely due to QScintilla's C++ parser tagging the whole line as a preprocessor directive.
The solution is probably to write our own parser component for QScintilla.

[image]

ChrisCoxArt, 2021-12-19:
The problem is in the scintilla CPP lexer used by Qscintilla.

QScintilla/scintilla/lexers/LexCPP.cpp LexerCPP::Lex()
```
			} else if (sc.ch == '#' && visibleChars == 0) {
				// Preprocessor commands are alone on their line
				sc.SetState(SCE_C_PREPROCESSOR|activitySet);
```
The logic after setting the preprocssor state is very C/C++ specific and not very customizable.
The rest of the state machine under 	
```
		case SCE_C_PREPROCESSOR:
```
is also veryC/C++ specific and non-customizable.

We can do a little better and highlight everything after the first word by setting 
        newLexer->setStylePreprocessor(true);
in the OpenSCAD code (ScintillaEditor.cc / void ScintillaEditor::setColormap() ). That will highlight everything after the first word, but the first word will remain unhighlighted until there is a change to the scintilla code.

t-paul, 2021-12-19:
For reference, the separate lexer code is in branch: https://github.com/openscad/openscad/tree/lexertl

ChrisCoxArt, 2021-12-19:
Relevant bits of the branch (which hasn't been updated since 2015):
qscintillaeditor.cpp
scadlexer.cpp
lex.cc
lex.h
lexertl/*

ChrisCoxArt, 2021-12-19:
The lexertl library could be added to the dependency downloads, but homebrew doesn't list it, which would break the homebrew builds.
I guess it could go into source/ext or somewhere in the openscad tree.

t-paul, 2021-12-19:
It's a not too big header only library, so integrating into src/ext should be ok. I guess we would need to switch to the C++14 version though, which seems to be maintained https://github.com/BenHanson/lexertl14.

ChrisCoxArt, 2021-12-19:
Do we want a partial fix for the time being (allow syntax coloring for everything after the first word after the '#'), or wait for a complete fix ?
The partial fix is one line
```
 newLexer->setStylePreprocessor(true);
```

t-paul, 2021-12-19:
If it helps a bit, why not. I would assume it can't make it worse than it is now. Does it also help with the auto completion on that line?

ChrisCoxArt, 2021-12-19:
OK, give me a few minutes to find a branch that isn't already waiting on a pull request.
As far as I can tell, auto completion is already working on # lines.

ChrisCoxArt, 2021-12-19:
and now we wait an hour for all the checks to run...

ChrisCoxArt, 2021-12-19:
All checks done after 103 minutes.
If I'd know that, I would have ordered dinner at the start...

ChrisCoxArt, 2021-12-19:
ok, got lexertl14 sorta working.  Biggest problem right now is that the categories in the current and lexertl descriptions don't match up at all.  I'll have to see what sort of temporary mapping I can come up with.

t-paul, 2021-12-19:
Yes, we could extend the color schemes with a dedicated new section which allows for the features lexertl can support an keep the other ones until we fully switch.

ChrisCoxArt, 2021-12-20:
I've gotten things much closer to working, and only really needed one new color category (variables).

What remains: getting the new lexer to load custom keyword sets from the json files (only one example uses them).  Do I try to replace the predefined language set, or add them as additional keyword sets?   The current code replaces the default set, which is not the best of ideas.

t-paul, 2021-12-20:
Adding new sounds like a safer way. Once all looks good with the new code, we can just remove the old ones.

ChrisCoxArt, 2021-12-21:
OK, the new lexer and colors are working well. What is the next step to get this tested?

t-paul, 2021-12-21:
Nothing special, just a normal pull request.

ChrisCoxArt, 2021-12-21:
Also, is it better to leave the old code commented out, just in case of a massive bug, then remove the old code later.  Or go ahead and remove the old code now, before testing?

t-paul, 2021-12-21:
I'd say it's fine either way, the old code is still in history too. But if you feel more comfortable to leave it in place for a bit, that's ok.

PR #4026 (2021-12-23): Lexertl integration - upgrade syntax coloring
Integrate lexertl lexer (header only code) to give us a custom lexer for syntax coloring, based on incomplete attempt in https://github.com/openscad/openscad/tree/lexertl/src with updated code from https://github.com/BenHanson/lexertl14
Fixes #1172

Add new categories for syntax colors, more options for custom syntax keywords and colors, and 3 new color schemes based loosely on colors in other editors.
closed 2021-12-23 by t-paul