openscad/openscad #1174: mirror and scale by negative invert normals, only CLI

laird, 2015-01-23:
I've found that when I do this:

if (prostheticHand<.5) {
    doEverything();
    }
else {
    mirror([1,0,0]) doEverything();
    }

or this:

scale([1-2*prostheticHand,1,1]) doEverything();

(where prostheticHand is either 0 or 1, and doEverything() imports and scales STL files). The source is in https://github.com/e-nable/Assembler/tree/dev/Assembly , but I think the above is the essence of the issue.

the un-mirrored version (or scaled by [1,1,1]) is fine, while the mirrored version (or scaled by [-1,1,1]) appears to have inverted normals, in addition to being mirrored left/right. Slicers seem to deal with it, but it's clearly wrong. And it causes some very odd behavior when I try to subtract from the object.

Sadly the good and bad generated STL files I am working with are too big to attach as 'images', but if necessary I can upload them and provide links. But it appears to happen for all files I've tried, so I'm hoping repo shouldn't be hard.

There's one interesting twist: it only happens from the command line. When I export from the GUI the files are fine. But if I render like this:

 /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -o ../testpalm5r.stl  -D prostheticHand=1  Assembly.scad the rendered STL is inverted as well as mirrored left/right (which it should be). Running -D prostheticHand=0 is normal.

I've duplicated in versions 2014.12.31 (on Mac desktop) and 2014.10.02 (on Linux).

So my question: is there some other way that I should be doing the above? Or is this a bug?

laird, 2015-01-23:
On the right is an imported STL. On the left is the same thing, mirrored. Note the awesome inverted surface.

[image]

kintel, 2015-01-23:
I didn't manage to reproduce by running the cmd-line you linked. Could you provide a walk-through of how to reproduce?

laird, 2015-01-23:
Check out https://github.com/e-nable/Assembler

Go into the directory Assembler/Assembler (which contains Assembler.scad).

The full command lines (on my Mac) are 

/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -o testpalm6r.stl -D label=\"u.e-nable.me/gs34q\"  -D Left1=0 -D Left2=0 -D Left3=0 -D Left4=0 -D Left5=0 -D Left6=0 -D Left7=0 -D Left8=61 -D Left9=0 -D Left10=0  -D Right1=0 -D Right2=0 -D Right3=0 -D Right4=0 -D Right5=0 -D Right6=0 -D Right7=0 -D Right8=61 -D Right9=0 -D Right10=0 -D part=2  -D prostheticHand=1 -D gauntletSelect=7 -D fingerSelect=1 -D palmSelect=1 -D Padding=5 Assembly.scad

/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -o testpalm6l.stl -D label=\"u.e-nable.me/gs34q\"  -D Left1=0 -D Left2=0 -D Left3=0 -D Left4=0 -D Left5=0 -D Left6=0 -D Left7=0 -D Left8=61 -D Left9=0 -D Left10=0  -D Right1=0 -D Right2=0 -D Right3=0 -D Right4=0 -D Right5=0 -D Right6=0 -D Right7=0 -D Right8=61 -D Right9=0 -D Right10=0 -D part=2  -D prostheticHand=0 -D gauntletSelect=7 -D fingerSelect=1 -D palmSelect=1 -D Padding=5 Assembly.scad

kintel, 2015-01-23:
Thanks - that helped. A minimal example would help even more, but I can at least reproduce it now.

laird, 2015-01-23:
Thanks. I'll extract a minimal case.

laird, 2015-01-24:
OK, here's a minimal example:

repro-invert.scad:

f="cube.stl";
// above file generated by cube(10); export as STL from GUI to "cube.stl" in the same directory
//mirror([1,0,0]) 
import(f);

Execute 

/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -o result-cube.stl repro-invert.scad 

and it generates a cube that's not mirrored and not inverted, for reference

Then uncomment the 'mirror' line and run again

/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -o result-cube-mirror.stl repro-invert.scad 

and it generates a mirrored, inverted cube.

Load the two into a slicer (Simplify3D in this example) and arrange them next to each other, and get:

[image]

laird, 2015-01-24:
Did more testing. It's not dependent on STL import.

Simply mirroring a cube generates one that's inverted.

mirror([1,0,0]) cube(10);

or 

scale([-1,1,1]) cube(10);

Run from command line as '/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -o mirror-cube.stl repro-invert.scad' gives inverted cube. So it's not related to the STL import. And it's the same whether mirrored or scaled by -1.

Note that this minimal case is consistently inverted from both the command line and when exported from the GUI. That's different from my first testing, but might be related to my updating to the new build 2015.1.23.

laird, 2015-01-24:
Just found #393 which sounds quite similar, mentioning that scaling by negative numbers turns objects inside out. It's closed now, as being fixed, but it doesn't seem to be.

laird, 2015-01-24:
For what it's worth, mirroring twice does re-invert the normals, so mirror([1,0,0]) mirror([1,0,0]) cube(10) is the same as cube(10).

closed 2015-01-25 by kintel

laird, 2015-01-25:
Awesome!

commit d6f0b7dc (2015-01-25): Flip faces when transforming using a mirroring transform. Fixes #1174