openscad/openscad #1175: Include git commit ID in Help->Library Info

kintel, 2015-01-23:
We have git commit ID in the status bar and about box, but not in Library Info. Since the latter is copied and pasted often, it makes sense to include it there as well.

ArchimedesPi, 2015-05-12:
So, I'm assuming you're talking about this part of the status bar:
[image]
I just built OpenSCAD and it looks like that, I'm not sure where the git commit id you're talking about is. Can you clarify @kintel?

t-paul, 2015-05-12:
This is about the text content of the "Help->Library Info" window. Looks like it's already done in https://github.com/openscad/openscad/commit/9ef1717459cd5e3d38b04c84bfc86c3f19ee9758

I'm getting;
```
OpenSCAD Version: 2015.05.04.nightly (git 89371f6)
System information: Linux 3.19.3-dali1 #3 SMP Sat Apr 11 22:47:52 CEST 2015 x86_64 Debian GNU/Linux stretch/sid 8 CPUs 15.59 GB RAM
Compiler: GCC "4.9.2" 64bit
```

closed 2015-05-12 by t-paul