openscad/openscad #1176: Scintilla - caret still active when editor pane not active.

MichaelAtOz, 2015-01-24:
When the editor window is not active, for example you click in the console pane, the text caret is still visible and flashing in the editor - meant to indicate that it is ready to have text typed there, however it is not, keystrokes are ignored until you click in the editor.
The simple editor works correctly.
2015.01.16.

hroncok, 2015-01-27:
I can reproduce this, it is very annoying :(

kintel, 2015-01-27:
I can reproduce too. I hope QScintilla lets us fix this :)

kintel, 2015-02-03:
This looks like a bug in QScintilla:
https://github.com/paralect/robomongo/blob/master/src/third-party/qscintilla/Qt4Qt5/qsciscintillabase.cpp#L345

They hide the Focus Out event from the scintilla component for mouse-originated focus events !?!
We might be able to work around it by actively intercepting the focus out event and either calling into scintilla manually or reissuing the event with the ```Qt::ActiveWindowFocusReason``` reason.

kintel, 2015-02-04:
This will be annoying to fix without patching QScintilla. I'm postponing this to later, unless someone have a good idea.