openscad/openscad #1177: Openscad devel crash

medvidec, 2015-01-24:
Hello,

I downloaded 2014.12.04 snapshot and run the openscad. I changed some settings (unfortunately I do not know which one), closed the application and next run I cannot start it.Today I tried 2015.01.16 snapshot and it is still the same. 2014.03 release however starts properly.

Can I somehow delete some settings which cause crash ? Or can I check some debug log ?

I get following crash:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	openscad.exe
  Application Version:	0.0.0.0
  Application Timestamp:	0be80c40
  Fault Module Name:	ig4icd64.dll
  Fault Module Version:	8.15.10.2025
  Fault Module Timestamp:	4b2bd484
  Exception Code:	c0000005
  Exception Offset:	000000000003fa08
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1029
  Additional Information 1:	1c01
  Additional Information 2:	1c010afbfb53ab94a109ad57ed372840
  Additional Information 3:	cf72
  Additional Information 4:	cf72b4974b6ddc06f128c8cb73e1eb13

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-US\erofflps.txt

t-paul, 2015-01-24:
The  ```ig4icd64.dll``` seems to be related to the graphics driver, can you make sure you have the latest driver installed?

The settings are stored in registry at ```HKEY_CURRENT_USER\Software\OpenSCAD``` (see [Qt documentation](http://qt-project.org/doc/qt-4.8/qsettings.html#locations-where-application-settings-are-stored) for details). Removing this should reset all settings to default state.

If it starts again, please add the output of the "Help->Library Info" dialog.

medvidec, 2015-01-24:
I deleted the settings and run OpenSCAD 2015.01.16.

I noticed following:
Welcome screen is shown. When I choose "New"  - openscad always crashes. When I open some model, openscad crashes in 70% probability. 

Library info is following:
OpenSCAD Version: 2015.01.16
System information: Windows 7 SP1 (v6.1.1.0) 4 CPUs 3.80 GB RAM
Compiler, build date: GCC "4.9.2" 64bit, Jan 16 2015
Boost version: 1_57
Eigen version: 3.2.2
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 5.4.0
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.42.1
Application Path: D:/optm/openscad-2015.01.16
Documents Path: C:\Users\vaclavpe\Documents
Resource Path: D:/optm/openscad-2015.01.16
User Library Path: C:/Users/vaclavpe/Documents/OpenSCAD/libraries
User Config Path: C:\Users\vaclavpe\AppData\Local/OpenSCAD
Backup Path: C:/Users/vaclavpe/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/vaclavpe/Documents/OpenSCAD/libraries
D:/optm/openscad-2015.01.16\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.11.0
OpenGL Version: 2.1.0 - Build 8.15.10.2025
GL Renderer: Intel(R) Graphics Media Accelerator HD
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: no
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: no

Používám QGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_ARB_color_buffer_float
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_shader
GL_ARB_half_float_pixel
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_shader_objects
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_transpose_matrix
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_object
GL_EXT_multi_draw_arrays
GL_EXT_packed_pixels
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_swizzle
GL_IBM_texture_mirrored_repeat
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_WIN_swap_hint

kintel, 2015-01-25:
@medvidec It looks like your graphics card driver is 5 years old, and an Intel driver, which is known to have had serious problems in the past. Could you try upgrading to the latest driver and trying again?

medvidec, 2015-01-25:
Hello,

I upgraded to version below and now openscad looks stable (another benefit is correct behaviour of one another retro computer emulator).

Thank you for help.

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.11.0
OpenGL Version: 2.1.0 - Build 8.15.10.2993
GL Renderer: Intel(R) HD Graphics
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

closed 2015-01-25 by medvidec