openscad/openscad #1183: Double import: ERROR: CGAL error in CGAL_Nef_polyhedron3()

hroncok, 2015-01-27:
I have a valid STL file and the following code:

    import("half.stl");
    mirror([1,0,0]) 
    import("half.stl");

F5 is fine, F6 generates no output and produces an error:

    ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326
    ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326

One standalone `import("half.stl");` works fine. Adding a `translate()` so the objects don't share a surface doesn't help.  Removing the `mirror()` and keeping 2 imports produces the same error.

Happens with multiple inputs, but not with all. One input can be found [here](https://churchyard.fedorapeople.org/solocarbon2_fixed.stl) (CC BY-NC-SA, gianteye), the other one I've been not granted a permission to upload publicly, but it can be sent privately.

Running on 0e90e3b5

KristianE86, 2016-03-09:
Did you find a workaround?

MichaelAtOz, 2016-03-09:
[image]

KristianE86, 2016-03-10:
So the stl was not valid after all?

MichaelAtOz, 2016-03-10:
That's what non-manifold means. Yes.

kintel, 2024-03-25:
Status as of 2024-03-24: This now works when using Manifold