openscad/openscad #1184: Intersecting STLs won't union, ERROR: CGAL error in CGALUtils::applyBinaryOperator union

hroncok, 2015-01-27:
[Bunny_Trouble_piece.stl](http://www.thingiverse.com/thing:7578/#files)

    import("Bunny_Trouble_piece.stl");
    import("Bunny_Trouble_piece.stl");

Produces:

    ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1102

The same happens with:

    import("Bunny_Trouble_piece.stl");
    cube(10);

The same happens when one STL is translated, but intersects with the other. Error is gone when the two STLs don't share a volume.

Running on 0e90e3b

hroncok, 2015-01-27:
What I see now, is that netfabb reports the file as valid, but in fact, it has intersecting triangles.

[image]

That might be the reason for this error. The problem here is: Is it possible to handle this file anyway? If no, shouldn't the ERROR message be more user friendly?

kintel, 2024-03-25:
Status as of 2024-03-24:
* bunny & cube works with both CGAL and Manifold
* Two bunnies: Still fails with CGAL, but works with Manifold