openscad/openscad #1186: Console highlight gets stuck

kintel, 2015-01-28:
From [mailing list](http://forum.openscad.org/Console-highlight-issue-tp11209p11252.html):

* Run fresh OpenSCAD
* Open something
* create a syntax error (do not save - so it autosaves a backup)
* F5 - gets error msg
* select any part or all of the first console line (above the error highlight) & part of the error msg.
* F5 again, whole console should be hightlighted
* Fix error
* F5 whole console highlighted.

hroncok, 2015-01-28:
I was about to report this, bacuse it happened to me while testing yesterday reported issues, but was later unable to reproduce this. I guess I didn't figure out it was the text selecting that was crucial. Glad you found it.

kintel, 2015-02-03:
From [QTextEdit docs](http://qt-project.org/doc/qt-4.8/qtextedit.html#append):
"Note: The new paragraph appended will have the same character format and block format as the current paragraph, determined by the position of the cursor."

Looks like we should reset the cursor to the end of the console before appending anything.
...in a thread-safe way.

commit 4ce1a8d9 (2015-02-04): Move cursor to end of console to avoid inheriting selected paragraph style. Fixes #1186
closed 2015-02-04 by kintel