openscad/openscad #1190: crash when exporting on OSX

falkorichter, 2015-01-28:
I´m trying to render this:
```
difference(){
	import("MKII_No_Case.stl", convexity=5);
	translate([76,26.8,0]){
		cube([13.6,10,33]);
	}
	translate([76,26.8,0]){
		translate([6.75,-5,0]){
		rotate(a=45){			
				cube([10,10,30]);
			}
		}
	}
}
```
where MKII_No_Case.stl can be downloaded from http://www.thingiverse.com/thing:507576/#files

I did a F5 and F6 before File->Export->STL

then the app crashes:
```
Process:         OpenSCAD [41773]
Path:            /opt/homebrew-cask/*/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:      org.openscad.OpenSCAD
Version:         2014.03 (2014.03.09)
Code Type:       X86-64 (Native)
Parent Process:  launchd [308]
Responsible:     OpenSCAD [41773]
User ID:         501

Date/Time:       2015-01-28 12:31:03.072 +0100
OS Version:      Mac OS X 10.9.5 (13F34)
Report Version:  11
Anonymous UUID:  F4B88AA6-2DF9-DF61-94EB-BACA91BA785B

Sleep/Wake UUID: 899A1015-FDAF-4C6C-8D5D-6F31B2DE9FBF

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000

Application Specific Information:
Performing @selector(qtDispatcherToQAction:) from sender NSMenuItem 0x6000000bcc20
Assertion failed: (px != 0), function operator->, file ../libraries/install/include/boost/smart_ptr/shared_ptr.hpp, line 653.
 

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff849cd866 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff8f6c135c pthread_kill + 92
2   libsystem_c.dylib             	0x00007fff8977fb1a abort + 125
3   libsystem_c.dylib             	0x00007fff897499bf __assert_rtn + 321
4   org.openscad.OpenSCAD         	0x0000000100154022 MainWindow::actionExportSTLorOFF(bool) + 1794
5   org.openscad.OpenSCAD         	0x00000001002e2c59 MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 601
6   QtCore                        	0x00000001019344fd QMetaObject::activate(QObject*, QMetaObject const*, int, void**) + 1693
7   QtGui                         	0x0000000100c92753 QAction::activate(QAction::ActionEvent) + 227
8   QtGui                         	0x0000000100c46cd4 -[QCocoaMenuLoader qtDispatcherToQAction:] + 68
9   com.apple.AppKit              	0x00007fff8e324260 -[NSApplication sendAction:to:from:] + 327
10  com.apple.AppKit              	0x00007fff8e33f1c8 -[NSMenuItem _corePerformAction] + 394
11  com.apple.AppKit              	0x00007fff8e33ef04 -[NSCarbonMenuImpl performActionWithHighlightingForItemAtIndex:] + 117
12  com.apple.AppKit              	0x00007fff8e38e40d -[NSMenu _internalPerformActionForItemAtIndex:] + 35
13  com.apple.AppKit              	0x00007fff8e38e289 -[NSCarbonMenuImpl _carbonCommandProcessEvent:handlerCallRef:] + 104
14  com.apple.AppKit              	0x00007fff8e334ff6 NSSLMMenuEventHandler + 716
15  com.apple.HIToolbox           	0x00007fff8717d1d4 DispatchEventToHandlers(EventTargetRec*, OpaqueEventRef*, HandlerCallRec*) + 892
16  com.apple.HIToolbox           	0x00007fff8717c787 SendEventToEventTargetInternal(OpaqueEventRef*, OpaqueEventTargetRef*, HandlerCallRec*) + 385
17  com.apple.HIToolbox           	0x00007fff87190880 SendEventToEventTarget + 40
18  com.apple.HIToolbox           	0x00007fff871c6640 SendHICommandEvent(unsigned int, HICommand const*, unsigned int, unsigned int, unsigned char, void const*, OpaqueEventTargetRef*, OpaqueEventTargetRef*, OpaqueEventRef**) + 420
19  com.apple.HIToolbox           	0x00007fff871f9228 SendMenuCommandWithContextAndModifiers + 59
20  com.apple.HIToolbox           	0x00007fff871f91d0 SendMenuItemSelectedEvent + 178
[... 316 lines cut ...]
__IMAGE                             528K
__LINKEDIT                         73.4M
__TEXT                            142.8M
__UNICODE                           544K
mapped file                       100.3M
shared memory                        68K
===========                      =======
TOTAL                             615.7M
TOTAL, minus reserved VM space    615.2M
 

Model: MacBookPro11,1, BootROM MBP111.0138.B07, 2 processors, Intel Core i5, 2.8 GHz, 16 GB, SMC 2.16f63
Graphics: Intel Iris, Intel Iris, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x112), Broadcom BCM43xx 1.0 (6.30.223.154.65)
Bluetooth: Version 4.2.7f3 14616, 3 services, 23 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500,28 GB
USB Device: Internal Memory Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.2
```

t-paul, 2015-01-28:
Can you verify if that happens with the snapshot version (= release candidate for the 2015.02 release) too (see http://www.openscad.org/downloads.html).

falkorichter, 2015-01-28:
It does not crash, but also it does not prompt me for a filename...
```
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 7
Geometry cache size in bytes: 227616
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 3 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds
Current top level object is empty.
Current top level object is empty.
```

falkorichter, 2015-01-28:
Getting this when opening and doing F5 + F6:
```
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 7
Geometry cache size in bytes: 227616
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 3 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds
Current top level object is empty.
Current top level object is empty.
```

falkorichter, 2015-01-28:
This error was not copied:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe_prev->is_border() || !internal::Plane_constructor::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate() File: /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 251
```

kintel, 2015-01-29:
This is a duplicate of #945: The STL file contains "zero triangles" which cannot be processed by OpenSCAD. Flipping edges in MeshLab fixes it.

closed 2015-01-29 by kintel

falkorichter, 2015-01-29:
Thank you. I guess the crash will be fixed in the next version.

kintel, 2015-01-29:
It won't be fixed in the upcoming version, but we'll work on it for the next one. Importing external STL files isn't OpenSCAD's big strength at the moment, but we'd love to make it a strength :)

As I mentioned, a workaround is to clean the file in an external program, like MeshLab.