openscad/openscad #1191: Add angle arguments to rotate_extrude()

clothbot, 2015-01-28:
Enhancement request, prompted by discussion http://forum.openscad.org/How-to-create-an-cylindrical-arc-td11257.html

Please add angle arguments to rotate_extrude() to support partial-arc extrusions:

Hypothetical usage:

module hollow_square(offset=10,a=45,w=5,th=1) translate([offset,0]) rotate(a) difference() {
  square(w,center=true);
  square(w-2*th,center=true);
}

// rotate_extrude from a=0 to a=30
rotate_extrude(a=30) hollow_square();
// rotate_extrude from a1=45 to a2=135
rotate_extrude(a1=45,a2=135) hollow_square();

kintel, 2015-01-28:
See also http://forum.openscad.org/partial-rotate-extrude-tt2695.html

We really only need one angle argument as it's trivial to add a rotate(). This also helps with orthogonality.

foulowl, 2015-04-14:
I would love to have this feature as well. Thank you!

thehans, 2015-11-21:
This can be closed since the merge of PR #1478

closed 2015-11-21 by t-paul