openscad/openscad #1193: CGAL Failure on Windows

TautologicalRecursion, 2015-01-29:
I'm really new to OpenSCAD, so I figured that I'd test myself with some basic projects to get a feel for the program. I tried today to model the sombrero function, and I finally did get it to work (that is, I could compile and preview it with F5...I'm missing some technical language I know). The issue is that when I tried to properly render the model to export it as an .stl (to 3D print), the progress bar stopped dead at 999/100 after fairly consistent progress, and though the program didn't crash, it didn't do much of anything either. I've attached an image which shows the code I used and the preview of the model. Any help is appreciated. 
[image]

MichaelAtOz, 2015-01-29:
At 999/1000 it is unioning all your objects, seems to be quite a few.

MichaelAtOz, 2015-01-29:
I've had complex objects that took hours to render, seconds to preview

TautologicalRecursion, 2015-01-29:
That's about what I figured...but with such a fast preview I expected at least something to happen. I'll let it run overnight and see what happens. (By the way, is there any easy way to increase the resolution of the shape? That is to say to increase the density of the polygons within the region rather than artificially bringing them closer with the x/2 and y/2 I used.)

kintel, 2015-01-29:
@TautologicalRecursion Not sure what you mean. To increase the density, use more cubes.

Perhaps a good extension for this sort of model would be to let surface() accept an array of values instead of just filenames.

MichaelAtOz, 2015-01-29:
I thing you want steps in the for(), for(x=[-20:0.5:20]), that will give you -20,-19.5,-19,-18.5 etc.

TakeItAndRun, 2015-01-29:
The way to get fast F6 on these things is to stitch  the surface by
polyhedron.
You can create the hole object in one go, thus avoiding the implicit union
of all your cube-columns at the end.

-- 
stempeldergeschichte@googlemail.com <karsten@rohrbach.de>

P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
schreiben.
Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.

P.S. In case my e-mail is shorter than you enjoy:
I am currently trying short replies instead of no replies at all.
Please let me know, if you like to read more.

Enjoy!

TautologicalRecursion, 2015-01-29:
@MichaelAtOz , that's exactly what I meant; thanks so much! 
@TakeItAndRun How exactly would I do that? It seems like that general process is the one being used to do the preview render (I think at least), but I'm not sure how to change the code to reflect that.

t-paul, 2015-01-29:
See https://gist.github.com/t-paul/d3f0ebe8ae97d1441f40 for an example.

TautologicalRecursion, 2015-01-30:
@t-paul Thanks for the great example! Something must be up with my list comprehension (I'm running windows, and I can't get even a simple list to operate correctly :/), so I can't check it myself :P

MichaelAtOz, 2015-01-30:
Works fine on Windows here.

TautologicalRecursion, 2015-01-30:
Gah. It seems that I get syntax errors with anything associated with requiring "2014.qx" on the Wiki. 
I have no idea how to proceed from here though...

MichaelAtOz, 2015-01-30:
Ah, you need the [development snapshot](http://www.openscad.org/downloads.html#snapshots)

TautologicalRecursion, 2015-01-30:
Ahh! Thanks! :D

kintel, 2018-12-07:
Not a bug

closed 2018-12-07 by kintel