openscad/openscad #1194: Font cache progress dialog crashe

kintel, 2015-01-29:
This was reported by quillford on irc:
o The font cache progress dialog shows for a minute or so, then crashes
o On the third attempt, it succeeded

It looks like a Qt threading issue:

```
Process:               OpenSCAD [27034]
Path:                  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:            org.openscad.OpenSCAD
Version:               2015.01 (2015.01.05)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenSCAD [27034]
User ID:               501

Date/Time:             2015-01-28 21:19:06.209 -0800
OS Version:            Mac OS X 10.10.2 (14C106a)
Report Version:        11
Anonymous UUID:        1E3012CD-30D9-28B6-8020-69879DE6CA6A

Sleep/Wake UUID:       822D3043-0FD3-4CE0-A3F7-9998E2F1D65C

Time Awake Since Boot: 120000 seconds
Time Since Wake:       13000 seconds

Crashed Thread:        10  QThread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
Pure virtual function called!
abort() called

Thread 0:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtGui          	0x00000001025f1ea5 qt_memfill32(unsigned int*, unsigned int, int) + 213
1   org.qt-project.QtGui          	0x00000001024ad0ec fillRect_normalized(QRect const&, QSpanData*, QRasterPaintEnginePrivate*) + 540
2   org.qt-project.QtGui          	0x00000001024ae8b7 QRasterPaintEngine::fillRect(QRectF const&, QSpanData*) + 247
3   org.qt-project.QtGui          	0x00000001024d1199 QPainter::fillRect(QRect const&, QBrush const&) + 393
4   org.qt-project.QtWidgets      	0x0000000101c26885 fillRegion(QPainter*, QRegion const&, QBrush const&) + 421
5   org.qt-project.QtWidgets      	0x0000000101c265be QWidgetPrivate::paintBackground(QPainter*, QRegion const&, int) const + 558
6   org.qt-project.QtWidgets      	0x0000000101c2dac9 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 2329
7   org.qt-project.QtWidgets      	0x0000000101c01a93 QWidgetBackingStore::doSync() + 4083
8   org.qt-project.QtWidgets      	0x0000000101c01dce QWidgetBackingStore::sync() + 510
9   org.qt-project.QtWidgets      	0x0000000101c24c0c QWidgetPrivate::syncBackingStore() + 236
10  org.qt-project.QtWidgets      	0x0000000101c504f3 QWidgetWindow::handleResizeEvent(QResizeEvent*) + 275
11  org.qt-project.QtWidgets      	0x0000000101c4ec64 QWidgetWindow::event(QEvent*) + 756
12  org.qt-project.QtWidgets      	0x0000000101bf505b QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
13  org.qt-project.QtWidgets      	0x0000000101bf83b0 QApplication::notify(QObject*, QEvent*) + 8192
14  org.qt-project.QtCore         	0x0000000102a2b6b3 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
15  org.qt-project.QtGui          	0x00000001022adb93 QGuiApplicationPrivate::processGeometryChangeEvent(QWindowSystemInterfacePrivate::GeometryChangeEvent*) + 403
16  org.qt-project.QtGui          	0x00000001022a9ef8 QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 88
17  org.qt-project.QtGui          	0x000000010229768a QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 346
18  org.qt-project.QtGui          	0x000000010229577f QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 543
19  libqcocoa.dylib               	0x0000000104ac7701 -[QNSView updateGeometry] + 545
20  com.apple.CoreFoundation      	0x00007fff8a4f3cdc __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
21  com.apple.CoreFoundation      	0x00007fff8a3e5244 _CFXNotificationPost + 3140
22  com.apple.AppKit              	0x00007fff8ffb2a99 -[NSView _postFrameChangeNotification] + 422
23  com.apple.AppKit              	0x00007fff8ffba80b -[NSView setFrameSize:] + 2059
24  com.apple.AppKit              	0x00007fff900777a0 -[NSWindow _oldPlaceWindow:] + 1627
25  com.apple.AppKit              	0x00007fff90076aba -[NSWindow _setFrameCommon:display:stashSize:] + 2527
26  com.apple.AppKit              	0x00007fff90867ceb -[NSWindow(NSWindowResizing) _resizeWithEvent:] + 1341
27  com.apple.AppKit              	0x00007fff902184f9 -[NSTitledFrame mouseDown:] + 200
28  com.apple.AppKit              	0x00007fff902182ea -[NSThemeFrame mouseDown:] + 68
29  com.apple.AppKit              	0x00007fff906b5178 -[NSWindow _reallySendEvent:] + 12721
30  com.apple.AppKit              	0x00007fff9013b87e -[NSWindow sendEvent:] + 446
31  libqcocoa.dylib               	0x0000000104abf467 -[QNSPanel superSendEvent:] + 39
32  libqcocoa.dylib               	0x0000000104abeb3b -[QNSWindowHelper handleWindowEvent:] + 411
33  com.apple.AppKit              	0x00007fff900edb61 -[NSApplication sendEvent:] + 4183
[... 492 lines cut ...]
MALLOC                            378.3M
MALLOC (admin)                       32K
Memory Tag 242                       12K
Memory Tag 251                       28K
OpenCL                               40K
OpenGL GLSL                         256K
STACK GUARD                        56.0M
Stack                              13.1M
Stack (reserved)                    520K        reserved VM address space (unallocated)
VM_ALLOCATE                        17.2M
__DATA                             26.6M
__GLSLBUILTINS                     2588K
__IMAGE                             528K
__LINKEDIT                         80.6M
__TEXT                            218.6M
__UNICODE                           544K
mapped file                       145.0M
shared memory                        68K
===========                      =======
TOTAL                             970.8M
TOTAL, minus reserved VM space    970.3M
 

 
```

kintel, 2015-02-03:
That didn't work too well - we'll try again..

closed 2015-02-03 by kintel

quillford, 2015-02-05:
All fixed. It didn't crash at all this time.

commit 8cb5adf0 (2015-02-03): Make sure GUI event processing happens in the GUI thread. Fixes #1194
commit 0fcc1fbf (2015-02-03): Force execution of the progress dialog to the GUI thread. Fixes #1194