openscad/openscad #1195: thread

kintel, 2015-01-29:


nophead, 2016-12-29:
> Currently, we perform Rendering (F6) in a separate thread, to ensure GUI responsiveness

The GUI is very very unresponsive during F6 on Windows. It isn't useable.

msrd0, 2016-12-29:
Can confirm GUI is responsive during rendering, only preview is not available, see #1901

MichaelAtOz, 2016-12-29:
> The GUI is very very unresponsive during F6 on Windows. It isn't useable.

That is not my general experience. 
[W7/64bit 4GB, SSD[with swap file], 2.54GHz x2 Core (ie 4 threads), cheap embedded graphics]
(I still haven't migrated to my beefier box, must get to that...)

However where a really large model + CGAL cache sizes are large*, and OpenSCAD process memory grows >~3GB swapping happens & it becomes bad. Prior to the SSD it was BAD, B A D.
I have a feeling in Windows 7, page faults get dispatched at kernel preemptive I/O priority which is above mouse/keyboard priority and GUI refresh.
I doubt that there is any application control on how virtual memory growth occurs on Windows.
(* I've had it at ~2GB - for no apparent reason other than big is better?)

--------------
Separate thought.
I don't know what is/not thread safe, but...
If every render() kicked off in a separate thread, given it is a sub-tree of the model, would that give us a chunky parallel option?

MichaelAtOz, 2016-12-30:
Should consider doing export in a thread as well. While it is shorter duration, for large models I frequently get the 'nonresponsive' label in the window title bar.

tkircher, 2023-11-04:
Still hoping for this to be implemented. Once in a while I would like to cancel a preview without using kill -9.

lgarron, 2025-06-16:
Once in a while I run into an issue where a relatively simple geometry change results in pathological changes in render times. (I'm talking a jump from about 1 second to over 10 minutes.)

It's sometimes possible to bring this back down by liberally sprinkling `render()` calls and using other tricks, but it's an extremely annoying process because I have to force quit and re-open the OpenSCAD app every time. It would be *extremely* helpful if the UI was responsive even during a long preview calculation.