openscad/openscad #1199: Feature request: Function scope and variables

oysteinkrog, 2015-02-01:
Currently it is possible to use variables in a function with this syntax (using let):

```
function func(arg1) = 
let(var1 = 1,     
     var2 = arg1)
var1+var2;
```

It would be nice if we could use the same syntax that is used in the rest of the language, e.g something like this:

```
function func(arg1) = 
{
     var1 = 1;
     var2 = arg1;
     var1+var2;
}
```

In the above version the return value is simply the last statement.
I know there are other issues/OEP's that cover this (e.g. OEP3), but this is a specific request for this small change (and only this).
For more complex functions this is sorely needed. Having to use let() seems unnecessary to me, especially considering the recent removal of assign().
For newcomers it would be a large improvement if the same syntax that is used everywhere else also worked inside functions.

kintel, 2015-02-01:
As this is directly mappable to a cascade of let() expressions, it should be doable.
Need to discuss a bit, but I _think_ I agree.

TakeItAndRun, 2015-02-01:
This looks like a syntacs  error. Maybe:

return=var1+var2;

-- 
stempeldergeschichte@googlemail.com <karsten@rohrbach.de>

P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
schreiben.
Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.

P.S. In case my e-mail is shorter than you enjoy:
I am currently trying short replies instead of no replies at all.
Please let me know, if you like to read more.

Enjoy!

MichaelAtOz, 2015-02-01:
That would make 'return' like a 'variable', I think a return statement may be better in this case.

However, alternatively, it could be the function name as a 'variable'.

Also, when using block mode, the '=' should be dropped.

Function func(arg1) {
  var1=1;
  var2=arg1;
  func=var1+var2;
}

??

oysteinkrog, 2015-02-01:
Personally I think we should stick with either "return var1+var2" or just "var1+var2" as the last statement, both of these alternatives are commonly seen in other languages, whereas using a special variable or something of that kind is a bit more obscure, and would also require the introduction of a new reserved keyword.

atartanian, 2015-02-02:
i really like the idea of having a 'return' keyword  and the ability to use block notation for functions as it really improves readability for complex functions.

MichaelAtOz, 2015-02-02:
I agree. 

Just pointing out that many new users, familiar with traditional programming, will expect other language features to work in the block, for(), if() [tho these can be used in list comprehensions], echo() [yaaah! fecho()], etc.

That could be a good thing. ? 

Then we could consider, for future evolution, whether geometry can be used in a block function. Returning geometry as a 'variable'.

function f(v=[1,1,1]) {
  cube(v);
}

p=f();
q=f([2,2,2]);
difference(q,f);