openscad/openscad #1202: 2D shapes with F6 don't render at z = 0

atartanian, 2015-02-02:
2D shapes rendered with F6 appear below the z = 0 plane
[image]

kintel, 2015-02-02:
I don't remember why we did that. If nobody remembers, we might just move them to 0 again :)

hroncok, 2015-02-02:
And the daemons will come back :D

MichaelAtOz, 2015-03-13:
A slight tangent, but perhaps this is also the time to make the F5 z-height/thickness 0.001 or such. Noting confusion that comes up in the forum from time to time.

TakeItAndRun, 2015-03-15:
+1

-- 
stempeldergeschichte@googlemail.com <karsten@rohrbach.de>

P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
schreiben.
Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.

P.S. In case my e-mail is shorter than you enjoy:
I am currently trying short replies instead of no replies at all.
Please let me know, if you like to read more.

Enjoy!

closed 2016-01-23 by kintel

MichaelAtOz, 2016-01-23:
> A slight tangent, but perhaps this is also the time to make the F5 z-height/thickness 0.001 or such. Noting confusion that comes up in the forum from time to time.

Should I raise another issue for this?

kintel, 2016-01-23:
@MichaelAtOz Please do - would be good to keep track of that.