openscad/openscad #1203: Window has no icon

hroncok, 2015-02-04:
The OpenSCAD main window has no icon and thus appears with default icon in window manager.

[image]

The most bottom window with default icon is OpenSCAD. Also appears this way with Alt+Tab. Should have the OpenSCAD logo/icon I guess.

kintel, 2015-02-04:
Could you describe your system and how you installed OpenSCAD (incl. what versions) ?

hroncok, 2015-02-04:
Oh, OK, sorry.

This is Fedora package from nightly builds. The same happens with latest release. Maybe it my packaging error.

kintel, 2015-02-04:
The icon works for the Ubuntu nightlies, so I'd investigate a Fedora packaging bug first.

hroncok, 2015-03-17:
Any tips how to debug this?

kintel, 2015-03-17:
I was of the impression that the application icon on Linux desktops are managed by the window manager, not the application itself, so it should get its application icon if the appropriate window manager config is installed, e.g. https://github.com/openscad/openscad/blob/master/icons/openscad.desktop.

t-paul, 2015-03-17:
Yes, I think the application icon should come from
`/usr/share/applications/openscad-nightly.desktop`

and the document icon from
`/usr/share/mime/packages/openscad-nightly.xml`

(paths from my Debian system using the nightly build)

kintel, 2015-03-17:
Note that the nightly stuff is new in master, after 2015.03.

hroncok, 2015-03-18:
Well, AFAIK this only affects the icon of app in menu (.desktop) and icon of .scad files (the mime file). Both works here.

hroncok, 2015-03-18:
So, I've just checked with GNOME and the icon is there. I think the problem is not in Fedora package but that OpenSCAD does not explicitly set the icon for the Window and expects the window manager will use the icon from the desktop file, which might work on GNOME or KDE, but does not work on Xfce.

kintel, 2015-03-18:
Oh, perhaps Xfce uses the "system tray icon" and defaults to nothing? http://qt-project.org/doc/qt-4.8-snapshot/qsystemtrayicon.html

hroncok, 2015-03-18:
Well, I don't think tray icon is what's there - tray is something completely different.

What about http://doc.qt.io/qt-4.8/qapplication.html#windowIcon-prop

kintel, 2015-03-18:
Could give this patch a try?
```
diff --git a/src/openscad.cc b/src/openscad.cc
index 33192ac..f00fc55 100644
--- a/src/openscad.cc
+++ b/src/openscad.cc
@@ -654,6 +654,7 @@ int gui(vector<string> &inputFiles, const fs::path &original_path, int argc, cha
 #else
        QTextCodec::setCodecForCStrings(QTextCodec::codecForName("UTF-8"));
 #endif
+ app.setWindowIcon(QIcon(":/icons/openscad.png"));
        
        // Other global settings
        qRegisterMetaType<shared_ptr<const Geometry> >();
```

hroncok, 2015-03-18:
Fixes it when running openscad from the source directory. Will check installed RPM after a while.

hroncok, 2015-03-18:
Yes, that fixes it.

closed 2015-03-21 by kintel