openscad/openscad #1206: Improve axis scale markers

kintel, 2015-02-05:
If the design is not centered in the origin and you zoom into it, the scale markers eventually disappear since you assume that the center of focus is the origin.
-> needs more testing.

MichaelAtOz, 2015-03-13:
I think the scale jumps up a level too soon.

A. at point just before they disappear
[image]

B. one zoom level out
[image]

jamessynge, 2019-01-26:
Are there any plans to work on the first issue listed (scale markers disappear)? I'm trying to use the scale markers to estimate (eyeball) a distance, but not at the origin, so the scale markers disappear as I zoom in.

kintel, 2019-01-26:
No immediate plans. This could be a great feature to work on for a new contributor.

erichstuder, 2026-11-15:
Working on it.
Have an idea. Looks promising.
Could take a while.

Let me know if someone else is already working on it.

Regards