openscad/openscad #1208: Graphics Glitches

samw3, 2015-02-06:
Hi.  I just downloaded OpenSCAD for the first time because I was wanting to remix this: http://www.thingiverse.com/thing:69261

I loaded it up and was met by horrible graphics glitches.  I'm pretty disappointed. :/

Is this a bug or is this just how OpenSCAD is?  It's gets even worse when your rotate it around; pieces disappear.

I'm on a Macbook Air with Mac OSX 10.10.2 (Yosemite), and am using OpenSCAD v2014.03  downloaded from the website.

[image]

[image]

kintel, 2015-02-06:
Such glitches in preview mode are normal if the design doesn't take care to properly make the subtracted parts a tiny bit larger than the positive part. In render mode (F6), this should not be an issue.
People are lazy and tend to choose to live with rendering artifacts rather than fixing the design.

Improving OpenSCAD to second guess designers is probably doable, but there are too many other things to improve for this to make it to the top of the list  : /

closed 2015-02-06 by kintel

samw3, 2015-02-06:
Ah, I see now.  The render view is slower but more accurate.  Ok, I can live with this.  Sorry for cluttering the issue tracker, but thanks for the response.

kintel, 2015-02-06:
If you spend time on the design, I'd try to identify the issues and fix them. It will save you a lot of time waiting for rendering, and be educational. Look up the # and % operators in the manual for helping hunt down the offending objects.