openscad/openscad #1209: Progress bar slightly too big, changes status bar height

MichaelAtOz, 2015-02-08:
Looks odd. Status bar grows in height while progress underway, then shrinks back down.
It seems to be the bar height, not the n/1000 text or stop icon.

ChrisVesters, 2017-06-01:
I looked into it, turns out that the status bar has height 23, while the progress bar has height 26.
It is not possible to shrink the progress bar, as this is fixed by Qt. The only option seems to be to make a bit more room for the status bar and increase it to 26, such that it can hold the progress bar.

ChrisVesters, 2017-06-01:
An alternative way is to set the max height of the statusbar to 23.
I have taken screenshots from before and after to see how the status bar looks. Only some whitespace before and after the actual progress bar is lost, so this is a viable option.

[image]
[image]

MichaelAtOz, 2017-06-01:
Looks good. I always like seeing a bit more usable screen real estate squeezed out, excessive white space is a pet hate. Just look at this page!

MichaelAtOz, 2017-06-04:
Fixed by #2050.
Thanks Chris. Claim the bounty.

ChrisVesters, 2017-06-07:
@kintel This issue can be closed after you do some checks yourself if you want.
Is there a procedure such that it is easier to close the issue?

MichaelAtOz, 2018-02-19:
Seems Windows version is different.

Using 2018.01.06 these show the height difference.
[image] [image]

MichaelAtOz, 2018-02-19:
I wonder if my Windows settings are at play?
i.e. 
[image]
Tho there is no 'status bar' option...

nophead, 2018-11-06:
What version of Windows is that, XP? It looks completely different for me on Win7.

I would add a picture but Github seems to have lost the ability at the moment.

MichaelAtOz, 2018-11-06:
That's Windows 7, Aero disabled.

nophead, 2018-11-06:
OK, same happens for me if I switch to "classic" and it also seems to be there when I switch back, but perhaps less size change because I had never noticed it before now.

t-paul, 2018-11-06:
On Linux with Hi-DPI (Dell XPS 13 @ 3840x2160) the status bar is too small.

[image]

It's not changing size with the progress indicator though.

[image]

MichaelAtOz, 2018-11-06:
> On Linux with Hi-DPI (Dell XPS 13 @ 3840x2160) 

I notice various OpenSCAD (& others) problems with Windows on my Ultra-Wide QHD  3440x1440 when I fiddled with the DPI.
But as I understand W7 isn't good at that, v's W10, so I haven't reported it.
Should I? (for the greater good, as I can live with it)

MichaelAtOz, 2018-11-07:
Side by side:
[image]