openscad/openscad #1210: Crash in CheckForRightSplice

charleskorn, 2015-02-08:
I am experiencing a crash issue on OS X when I go to export certain models to STL. Based on a `git bisect`, the issue was first introduced in 39d68238725749b79842d1da3c997913986b4992, and the issue is still present as of fe4997b38cf799b7cddb394425ddf717e3f3f5c5.

OS X crash log and sample file available from https://gist.github.com/charleskorn/7b640b14631c6cef344c (changing `orientation=TOWARDS_LEFT` to `orientation=TOWARDS_RIGHT` at the end of the file allows the file to be exported without crashing). 

Info from Help > LIbrary info:

```
OpenSCAD Version: 2015.02.08
System information: Mac OS X Version 10.10.2 (Build 14C109) 4 CPUs 16.00 GB RAM
Compiler, build date: Clang "6.0 (clang-600.0.56)" 64bit, Feb 8 2015
Boost version: 1_54
Eigen version: 3.2.1
CGAL version, kernels: 4.4, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.3.1
QScintilla version: 2.8.4
MingW build: No
GLib version: 2.40.0
Application Path: /Users/ckorn/<path to repo clone>/openscad/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/ckorn/Documents
Resource Path: /Users/ckorn/<path to repo clone>/openscad
User Library Path: /Users/ckorn/Documents/OpenSCAD/libraries
User Config Path: /Users/ckorn/Library/Application Support/OpenSCAD
Backup Path: /Users/ckorn/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/Users/ckorn/Documents/OpenSCAD/libraries
/Users/ckorn/<path to repo clone>/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.10.0
OpenGL Version: 2.1 INTEL-10.2.46
GL Renderer: Intel Iris Pro OpenGL Engine
GL Vendor: Intel Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
```

kintel, 2015-02-09:
Thanks for the report. I can reproduce it and will look into it.

kintel, 2015-02-09:
Reported bug in external lib: https://github.com/memononen/libtess2/issues/14

kintel, 2015-02-09:
@charleskorn Please give it a spin now

closed 2015-02-09 by kintel

charleskorn, 2015-02-10:
@kintel Yep, looks like it is working now. Thanks for your quick response.

commit c50d9272 (2015-02-09): Fixed bug in libtess. Fixes #1210