openscad/openscad #1211: Windows: Touch-based navigation broken in dev snapshots

kintel, 2015-02-08:
From [mailing list](http://forum.openscad.org/Touch-control-missing-in-DEV-builds-tt11546.html):

Touch-based navigation partially worked in 2014.03, but doesn't at all in dev snapshots.
Perhaps a Qt 5 issue?

TDeagan, 2015-02-08:
I've got a win 8.1 touchscreen laptop.  I'll see if I can replicate.

TDeagan, 2015-02-08:
On my Windows 8.1 64 bit Acer 111P-43BC Laptop;
Dev Build 2015.02.05, 64-bit responds to touch as if it were a right mouse click and drag event.

This is the same behavior as 2014.03.

Next question would be if this is the same on a 32-bit system.  I can try setting one up in VirtualBox, but I've had mixed results with the touch screen so I'm not sure how valid the results would be.

ukian, 2015-02-08:
Ok, I now can't reproduce it, touch is working as it previously did. It didn't work on two days I used it on the train last week, so there is/was something intermittent. I'll do some experiments and re-report it if I can get it to fail reproducibly. 

Cheers 
Ian

TDeagan, 2015-02-09:
No problem.  I've been dealing with gremlins like that all week, so I can totally relate.

closed 2015-02-14 by kintel