openscad/openscad #1212: 2015.02.05 dev 64-bit openscad.com triggering Avast Win64:Evo-gen [Susp] (Frequent false positive)

TDeagan, 2015-02-08:
When downloading 64 bit versions of the 2015.02.05 dev build on my win 7 64-bit system, I am getting an alert from my virus protection Avast.  The openscad.com file is moved to quarantine based on Win64:Evo-gen [Susp].

[image]

MalwareBytes and Windows Defender do not find a problem with the file.

This is a known frequent false positive issue with Avast.  I have submitted the file to their virus lab with a request to exclude.  

Additionally, Avast's web shield is blocking my access to the file once the alert is caught, so it looks like the download is broken from that point forward (no notification that it's Avast that's blocking, I have to turn off protection to download it.)

Not much we can do, wasn't sure if this should be an issue, but it will cause heartburn for some segment of the users, so I thought I'd document.

t-paul, 2015-02-08:
Yes, that was reported against OpenSCAD-2015.01.23-x86-32.zip before. It's likely caused by the registry poking. Very bad heuristics, essentially no other scanner triggered when trying some of the online scanner services.

kintel, 2015-02-09:
I was also suspecting the registry modifications as I got this reported shortly after merging the icon pull request. Should be easy enough to test, but commenting out that code and trying the scanner again..