openscad/openscad #1213: Dependecies to build OpenSCAD from sources in Ubuntu 14.04 amd64

SEUTec, 2015-02-09:
Hi all:

I have build OpenSCAD from the sources in github.
But I have found a dependences that is not detected by the script uni_get_dependecies.sh and/or check-dependecies.sh.

While compiling is missing the fiile Qsci/qscintilla.h

I have solved this problem installing libqscintilla2-dev and its dependencies.
After that. All seem to wokr fine. Compiling and running new OpenSCAD version 2015.02.09.

Many Thanks, for your very interesting and usefull work in OpenSCAD.

Regards.

SEUTec.
http://usuaris.tinet.cat/sje/index_en.htm

t-paul, 2015-02-09:
The ```uni-get-dependencies.sh``` script installs Qt5 and libqt5scintilla2-dev. You probably have Qt4 too and did build against that (which is fine).

Unfortunately the parallel support for Qt4 and Qt5 which is different for almost any Linux distribution makes it not especially easy to document the build process for all variations.

SEUTec, 2015-02-10:
Ok. Thanks.

closed 2015-02-10 by SEUTec