openscad/openscad #1217: Sequential assignment of default module arguments

daprice, 2015-02-15:
It would be helpful if default arguments for modules could be defined dyamically and sequentially, so each definition could use the value of any previous arguments (similar to how the proposed `let()` function works). For example, in the following code, the assignment of `b` should be able to use the value of `a`:
```
module myModule(a, b=a) {
	echo(b);
}

myModule(a=1);
```

If this was implemented, the echoed output would be `1`.

However, in the current nightly this just causes a warning: `Ignoring unknown variable 'a'.`

MichaelAtOz, 2015-02-16:
Then

s=0.01; // smidgen

module m(params,s=s*2,s2=s*4) {
  echo(s,s2);
}

Would not work anymore.

MichaelAtOz, 2015-02-16:
oops markdown mishap, make that `s=s*2,s2=s*4`

kintel, 2016-03-03:
I agree with @MichaelAtOz - this would break compatibility of such a corner case.
We do plan to break such compatibility in the future, so that parameter lists behave the same way in all context. When we do, we will provide either a compatibility mode or an upgrade option for old scripts.
That's a bit of work, so we'll probably switch it all at the same time.

doug-moen, 2016-04-09:
@kintel said "so that parameter lists behave the same way in all contexts"

I can see the use case for default values in function and module definitions. But in function and module calls, it's less clear that we want "sequential assignment".

Here's some code that was just posted to the forum:
```
function merge(left, right, kp, kl=0, kr=0, _acc=[]) =
    kl >= len(left) && kr>= len(right) ? 
        _acc:
        merge( left  = left ,
               right = right ,
               kp    = kp,
               kl  = next_k(kl, kr, len(left),  len(right), kp, left[kl], right[kr]),
               kr  = next_k(kr, kl, len(right), len(left),  kp, right[kr], left[kl]),
               _acc  = concat(  _acc, 
                                next_entry(kr, kl, len(left), len(right), kp, left[kl], right[kr] )
                             )
             );
```
Look at the recursive call to merge(), which sets the kl= and kr= parameters based on the values of kl and kr in the parent invocation merge(). This code would be harder to write if function calls were changed to use sequential assignment for keyword arguments.

kintel, 2016-04-09:
@doug-moen Point taken. Any ideas of a pattern to follow, or how to communicate this behavior clearly?

doug-moen, 2016-04-10:
@kintel If you accept my premise that function call labelled arguments shouldn't have a scope, then here are two reasons for leaving the semantics of function definition formal parameter lists the way they are:

1. It will be consistent with the way function calls work.
2. It's the easiest way to fix #1594. In OpenSCAD2, functions are values, so function definitions and function literals need to be evaluated to produce a function value. We'll just say that default values in parameter lists are evaluated at the time that the function is evaluated, and not at the time the function is called. That fixes #1594.

The down side is that it is more work to specify formal parameters whose default values depend on other formal parameters. These parameters will need to be specified with a default value of `undef`, and the work of computing the parameter's actual value needs to occur in the function body.

The idiom looks like this:
```
module myModule(a, b=undef) {
    actual_b = b==undef ? a : b;
    echo(actual_b);
}
```
Since it's already possible to write this code in OpenSCAD, what @daprice is requesting is a short cut, not a new language ability.

If you consider the general case of "dependent parameters" (whose default values depend on other parameters), then Dale Price's short cut only works for special cases, not the general case. I remember seeing code where there are two parameters, each having a default value that depends on the other parameter (sorry I can't find the link right now). In OpenSCAD2, functions will be passed as arguments, and some function parameters will have default values. There are examples of this in ImplicitCAD. I can imagine a default function value that depends on other parameters, is recursive, and depends on auxiliary functions that also depend on parameters. That's kind of a worst case scenario, and it's clear that such a default value would have to be constructed in the function body, not in the parameter list.