openscad/openscad #1218: rotate_extrude With Rotation about Y-axis

SuperContraptionGuy, 2015-02-16:
It doesn't seem possible to rotate a profile about the y-axis prior to a rotate_extrude command, it doesn't effect the output.
```OpenSCAD
x = 45;

rotate_extrude()
	#translate([3, 0, 0])
		rotate([90, x, 0])    //Rotates first on the x-axis 90* to create an 'extrudable' profile, then it attempts to rotate arount the y-axis.
			polygon([[0, 0], [1, 0], [0, 1]]);    //simple triangle
```

The output of the above looks like the picture below, regardless of what the value of 'x' is:

[image]

The  highlighted object is the profile to be extruded.  As you can see, it is successfully rotated, but the rotate_extrude function disregards the rotation and extrudes the original profile.

MichaelAtOz, 2015-02-16:
rotate_extrude operates on 2D objects, rotating around x or y results moves the object out of the 2D plane. rotate_extrude basically does the rotation for you.

SuperContraptionGuy, 2015-02-16:
The object is still on the X-Z plane even after a Y rotation though.

MichaelAtOz, 2015-02-16:
If you rotate on Y (or X) you rotate out of the XY 2D plane into 3D

You're probably seeing the infamous 1mm high 2d objects and getting confused.

I change the scale by 100

```
x = 45;

rotate_extrude()
    !translate([300, 0, 0])
        rotate([90, x, 0])    //Rotates first on the x-axis 90* to create an 'extrudable' profile, then it attempts to rotate arount the y-axis.
            polygon([[0, 0], [100, 0], [0, 100]]);    //simple triangle
```
Note the '!' which shows that the triangle is rotated into 3D.

MichaelAtOz, 2015-02-16:
Perhaps, again, it is time to set preview to 0.01mm high??????

SuperContraptionGuy, 2015-02-16:
I figured out how to do what I want, but it is weird how it didn't work before.

For some reason, you have to rotate about first the z, then the x, instead of the x then the y.  Both cases result in the same translation... Annoying...

Should it be considered a bug? why? why not? I'm curious.

```
x = 45;

rotate_extrude()
    #translate([300, 0, 0])
        rotate([90, 0, 0]) //Then the X-axis
            rotate([0, 0, x]) //First the Z-axis
                 polygon([[0, 0], [100, 0], [0, 100]]);    //simple triangle
```
[image]

MichaelAtOz, 2015-02-16:
What version of OpenSCAD are you using?

If I have the `rotate([90, 0, 0]) //Then the X-axis` line it produces a flat washer
It is not valid.

If I comment it out I get the above.

SuperContraptionGuy, 2015-02-16:
Hmm... strange...
I'm running 2014.03 Linux.
Without the ```rotate([90, 0, 0]);``` Then the polygon is on the XY plane, it won't extrude.  It has to be in the XZ plane to work with ```rotate_extrude()``` .  That's what I've known anyway, has worked for me every time.

SuperContraptionGuy, 2015-02-16:
Actually, it seems to work wether I have the ```rotate([90, 0, 0]);``` or not... Now I'm a little confused.

MichaelAtOz, 2015-02-16:
Just checked with 2014.03 (Win). It does the same, ie works.

2015.02.05, latest development version - about to be released, does not support the rotation into X or Y.

Check the [manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/2D_to_3D_Extrusion#Rotate_Extrude) the 2D object must be on the XY plane and should be entirely in the positive quadrant.

SuperContraptionGuy, 2015-02-16:
I'll keep that in mind for compatibility's sake. Weird...

kintel, 2015-02-16:
On Feb 16, 2015, at 00:16 AM, SuperContraptionGuy <notifications@github.com> wrote:

> The object is still on the X-Z plane even after a Y rotation though.
> 
If you want to rotate the resulting object, just apply rotation on top of the rotate_extude:
rotate() rotate_extrude() 2d_object();

 -Marius

kintel, 2015-02-16:
Wops, missed the rest of the discussion.
rotate_extrude requires a 2D polygon, meaning that only the XY coordinates of passed geometry will be taken into account.
Recently, OpenSCAD has started to allow application of 3D transformations to 2D polygons, without clarifying its limitations.
I guess a reasonable behavior would have been to continue to ignore 3D transformations until we have a good workflow for those.

We�re close to releasing a new version. If you have any strong wishes for changes, this would be a good time..

 -Marius

atartanian, 2015-02-16:
This is definitely a source of confusion for new users which I've run into
a number of times while teaching openscad. That being said, I would be sad
to lose the ability to visually transform 2D shapes around in 3D. I use
that "quirk" to be able to efficiently preview geometry in customizer
without actually calculating the 3D shape. Personally, it seems like the
quick fix would be for openscad to throw a warning (but should still
generate the same visual output) in the output console if it sees you
trying to transform a 2D shape into a 3D context (rotate around X or Y or
translate in Z). More specifically, I think the main confusion with
rotate_extrude is that there is a hidden rotation around the x axis that
isn't intuitive. it would be easier to understand if it just did the
rotational extrude around the y axis and left the object like that (for
instance an upright torus). I would really love it if 2D shapes kept track
of their own local coordinate system, that was used to do extrusions. That
way, even if the shape was transformed in the global space, the extrusion
would be based on the shape's local orientation. The example above would
work as expected if that were the case... (as long as the arbitrary x axis
rotation was removed as previously mentioned)

MichaelAtOz, 2015-02-17:
> That being said, I would be sad to lose the ability to visually transform 2D shapes around in 3D. I use that "quirk" to be able to efficiently preview geometry in customizer without actually calculating the 3D shape.

You will be sad. With 2015.02.05 if it is rotated. STOP the press.
I was about to say if rotated around X it doesn't work.
But I tested further.
If rotated([90,0,0]) it does a flat washer.
However any lesser number and it produces a 3D object....

I think a warning may be necessary.

atartanian, 2015-02-17:
@michael oh yeah I wasn't talking about rotate_extrude. I was responding to
marius saying "I guess a reasonable behavior would have been to continue to
ignore 3D transformations". I just meant that I would be disappointed if 2D
shapes lost the ability to LOOK like they had been moved around in 3D in
PREVIEW, since I use that (along with scaling them in the z axis) to
simulate a linear_extrusion without the cost of actually creating the 3D
shape (since extrusion is done by CGAL if i remember correctly).

That being said, I acknowledge that the behaviour of allowing 2D shapes to
LOOK transformed in 3D but not allow operations to use that new orientation
is definitely confusing. So I'm between a rock and a hard spot, since I
desire a clean and intuitive UX but also have a non-standard use for a
quirky behaviour. :(

t-paul, 2019-11-22:
Behavior of `rotate_extrude()` is probably a bit unexpected when using the first time, but that's not easily changeable. There's now an extensive [tutorial](http://files.openscad.org/tutorial/) being written which has a section explaining that.

closed 2019-11-22 by t-paul