openscad/openscad #1219: difference and intersection don't work in Windows 10 Pro technical preview (build 9926) !

lucrol, 2015-02-16:
This week-end I try to use OpenSCAD version 2014.03 on my ultrabook (Acer Aspire S7-391, Intel  core i7-3517U, RAM 4 Go, video Intel Graphics HD4000, SSD 256 Go) running the Windows 10 Pro technical preview (build 9926).
The difference and intersection statements don't work : no display
Same code works fine on my Workstation (Windows 8.1U1  enterprise) ...

Luc

t-paul, 2015-02-16:
Can you please try the DEV Snapshot from http://files.openscad.org/snapshots/OpenSCAD-2015.02.15-x86-64.zip (That's Release Candidate 3 for the next release) and post the output of the "Help -> Library Info" Dialog.

The problem is likely the graphics driver, the info dialog should list some details about the OpenGL driver.

lucrol, 2015-02-20:
Thanks Paul
This Release Candidate 3 feel great, but I've the same problem using Windows 10 Technical preview.
I join the images containing the output of Help / Library info.

Luc
[image]
[image]
[image]
[image]
[image]

kintel, 2015-03-03:
@lucrol Windows previews tend to have problem with OpenGL. Could you try the following:
1) Upgrade to OpenSCAD-2015.02.25
2) Upgrade your OpenGL drivers to the latest version

lucrol, 2015-03-09:
Hello,
I do the upgrade so :
- Simple use of difference and union works now,
- but nested difference don't.
OpenGL drivers are Embedded in Win 10, 
No specific video driver for this preview from cer or intel.
Windows Update is up to date.
Many thanks.

Luc

kintel, 2015-03-09:
@lucrol Have you tried to update the drivers yourself? https://downloadcenter.intel.com/download/24593/Intel-HD-Graphics-Driver-for-Windows-7-8-8-1-64-bit

t-paul, 2016-09-20:
Please check again with latest snapshot builds. Updated OpenCSG now uses different rendering that might help with Intel drivers (see #1409).

closed 2019-01-25 by t-paul