openscad/openscad #1224: Feature Request: Path Sweeping?

eddieparker, 2015-02-23:
I would love if I could define a path and sweep an object along it to create the shape.  

It's easier (for me, anyways) to define an object based upon the path it sweeps and a thickness of the object, then it is to remember the winding order and adjust triangle faces.

I could see something like this being awesome:

path_sweep(points=[[0,0,0], [0,1,0], [1,0,0]])  // Generate a path
cube([1,1,1]*0.5); // Make this cube sweep along it.

I could likely implement something like this by using a for loop of tight enough constraints, but i'm hoping a language feature with this could do this more optimally, with nicer syntax.

Any chance this exists by another name? :)

NateTG, 2015-02-23:
Look at:

https://github.com/openscad/openscad/issues/114

There are sweep and skin userspace functions.

I'm not sure anyone's done the sweep of a 3D object in that thread.

Ghoughpteighbteau, 2015-03-04:
I implemented something like this, if you're interested @eddieparker.

I was making these marble blocks and used a sphere along a path to subtract the path of a marble from out of a block.

It's not particularly efficient, but it does sweep an arbitrary path. We'll have to wait for 114 before anything easier comes up. 

```
//Quality
carveSteps = 8; //20-30 is high quality, 6-8 is low
$fa = 20;         //5-8 is high quality, 20-30 is low

//Bezier functions
function bez1(p1, p2, t)         = p1*(1-t)               + p2*t;
function bez2(p1, p2, p3, t)     = bez1(p1,p2,t)*(1-t)    + bez1(p2,p3,t)*t;
function bez3(p1, p2, p3, p4, t) = bez2(p1,p2,p3,t)*(1-t) + bez2(p2,p3,p4,t)*t;

module bezier2(p1, p2, p3, step){
	translate(
	[bez2(p1[0],p2[0],p3[0],step)
		,bez2(p1[1],p2[1],p3[1],step)
		,bez2(p1[2],p2[2],p3[2],step)
		]) children();
}
module bezier3(p1, p2, p3, p4, step){
	translate(
		[bez3(p1[0],p2[0],p3[0],p4[0],step)
		,bez3(p1[1],p2[1],p3[1],p4[1],step)
		,bez3(p1[2],p2[2],p3[2],p4[2],step)
		]) children();
}
module drawBezier2(p1, p2, p3){
	for(i = [0 : 1 : carveSteps-1]) hull(){
		bezier2(p1,p2,p3,i/carveSteps) children();
		bezier2(p1,p2,p3,(i+1)/carveSteps) children();
	}
}
module drawBezier3(p1, p2, p3, p4){
	for(i = [0 : 1 : carveSteps-1]) hull(){
		bezier3(p1,p2,p3,p4,i/carveSteps) children();
		bezier3(p1,p2,p3,p4,(i+1)/carveSteps) children();
	}
}

drawBezier3([50,0,50],[0,0,50],[0,0,0],[0,50,0]) sphere(16);
```

[image]

I'm sure there's a clever variadic recursive implementation of bezier curves that can handle any order, but this does the job. 

Oh, I also didn't spend the time working out the orientation of an object. If you want that, you'll have to do some simple trig on `bezier(...,i)` and `bezier (...,i+1)`

hope this helps!

jordanbrown0, 2025-09-01:
Note that #114 contemplates sweeping a 2D object along a path (and so linear extrusion would be a subset, and rotational extrusion a subset if closed paths are supported), but this one contemplates sweeping a 3D object.

tcurdt, 2026-08-30:
unstale