openscad/openscad #1228: OpenSCAD exits when the last window is closed

Feuermurmel, 2015-02-26:
At least on OS X, when the last window is closed, the application exits. I assume that this behavior is present on all platforms but at least on OS X it is contrary to the conventions of document-based applications.

An even bigger issue IMHO with this behavior is that all cached results need to be re-calculated.

kintel, 2015-02-26:
The reason for this is that the menus are attached to each Window, and if we leave the application open, we don't have a way of opening new windows.
One workaround could be to bring back the splash screen when the last window is closed.

..or rewrite the GUI to handle MDI better on Mac. That's pretty annoying to do in Qt though..

kintel, 2024-12-03:
Closing as not planned

closed 2024-12-03 by kintel