openscad/openscad #1230: The same file can be opened multiple times

Feuermurmel, 2015-02-26:
This may have specifically implemented as a feature, so forgive me if that's the case. If a file is opened (either via _File_ > _Open…_ or by opening the file from the Finder) that has already been opened, a new window for the same fie is opened. I would have expected that this would just bring the existing window for that file to the front.

kintel, 2015-02-26:
TODO: Define correct behavior for each platform

MichaelAtOz, 2015-02-26:
If this is implemented there should be a way to open the same file in more than one window/and/or/instance. I do this from time to time.

kintel, 2015-02-26:
Right, I guess a common use-case is to render different views of the same design.

MichaelAtOz, 2015-02-26:
Yes, with auto-reload, it works nicely.

Feuermurmel, 2015-02-26:
@MichaelAtOz I didn't think about that! I agree that this is useful and I know a lot of programs that offer that functionality. A common implementation is to provide a command along the lines of "New Window for <document name>" in the _View_ menu. Maybe even add a button to the toolbar.

jotpandher, 2015-04-23:
Done.