openscad/openscad #1232: Preview is not automatically updated when imported files change

Feuermurmel, 2015-03-01:
When _Design_ > _Automatic Reload and Preview_ is activated, the preview is automatically updated when the opened `.scad` file or other files included using `include <...>` or `use <...>` are changed. This does not work for files imported using `import(...)`. I'm observing this on __OS X 10.9.5__ with version __2015.02__.

I'd claim that it would be useful if the preview was automatically updated when such a file changes.

kintel, 2015-03-01:
It's not yet part of the automatic reload feature, but I agree that it would be useful when editing imported files in parallel with creating the OpenSCAD design.

jordanbrown0, 2025-03-09:
Applies to `surface()` too.