openscad/openscad #1234: openscad uses an ineffecient union technique

Ghoughpteighbteau, 2015-03-03:
This is technically a problem with CGAL. It **SHOULD** be using a divide and conquer strategy, what it's currently doing (or what openSCAD is doing, I need to actually check the source to see what's going on) is a naive way of working.

[PostGIS did the same exact thing, but they fixed it](http://blog.cleverelephant.ca/2009/01/must-faster-unions-in-postgis-14.html)

Here's a demonstration of the problem:
```
n=3;
$fa=40;

!for(x=[0:n],y=[0:n],z=[0:n])
translate([x,y,z]*10) sphere(15);

for(x=[0:n]) for(y=[0:n]) for(z=[0:n])
translate([x,y,z]*10) sphere(15);
```
for me, running this code takes this long:
```
Total rendering time: 0 hours, 1 minutes, 11 seconds
Rendering finished.
```
but running the second statement takes:

```
Total rendering time: 0 hours, 0 minutes, 28 seconds
Rendering finished.
```

The reason for the decreased time is that each union you perform is O(n*m) minimum for each vertex in the operation (it may be worse than that...) So by unioning the smallest objects in the set together, you eliminate verts as quickly as possible, and reduce workload. 

The higher `n` is set to in this program, the more extreme the difference in speed. OpenSCAD could easily improve the speed of unions with many objects by implementing a similar strategy.

For example, this union has considerably better speed than a native union on large sets:
```
runion(){
	if($children > 2){
		runion() children([0:2:$children-1);
		runion() children([1:2:$children-1);
	}
	children();
}
```
I'd be using this in my project right now, but because for loops implicitly union, I can't!

t-paul, 2015-03-03:
Improving the implicit union is tracked in https://github.com/openscad/openscad/issues/350 - the hardest problem with that is to remain backward compatible with existing scripts.

nophead, 2015-03-03:
So what happens if you have a missing union in an old script. Does it
result in overlapping volumes and a non-manifold error?

kintel, 2015-03-03:
@nophead Overlapping volumes is usually the correct result. Most slicers should handle that correctly (i.e. perform the implicit union at slice-time but in 2D so it's fast).

One of the reasons for not merging this is that we haven't figured out how to best ensure backwards compatibility. The [Unwrap Operator](https://github.com/openscad/openscad/wiki/OEP4%3A-Unwrap-Operator) has been suggested.

nophead, 2015-03-03:
Have a function to allow the script to declare which version of OpenScad it
was written for. If not called assume before this change. Then you can
always have backwards compatibility although it involves keeping legacy
code.

kintel, 2015-03-03:
In terms of versioning, we've been considering adding an optional header: https://github.com/openscad/openscad/wiki/File-Format-Versioning#versioning

Ghoughpteighbteau, 2015-03-03:
wait wait wait. Yes adding lazy union would be great, but it wouldn't actually solve the problem this issue describes. The `runion` module I described is a very *inefficient* improvement over an already naive algorithm.

what I believe to be ideal behavior isn't possible in userland because a balance between two factors needs to be reached:

1. you must union shapes with the least number of vertices first. 
2. you must union shapes that eliminate vertices as soon as possible.

This change would be 100% backwards compatible. 

Ferreting out the optimal algorithm might be worth some time, but even something simple would show massive speed improvements.

kintel, 2015-03-03:
@Ghoughpteighbteau In OpenSCAD, we use CGAL's ```Nef_nary_union_3```, which hints at how to schedule the objects in a more efficient way. They suggest adding polyhedrons in "sorted order", not using a divide and conquer mechanism. Does this make any sense to you? http://doc.cgal.org/latest/Nef_3/classCGAL_1_1Nef__nary__union__3.html

Ghoughpteighbteau, 2015-03-03:
Yes.

It seems my statement that CGAL use a divide and conquer strategy was a little overzealous. They have made a speed/memory tradeoff. By having a running aggregate, they need only store one intermediary product. If you provide your objects in sorted order you achieve the second factor of a good algorithm, eliminating vertices as quickly as possible.

In the example I provided when opening the issue, the objects provided to CGAL *are* entered in sorted order, verticies are eliminated every step, but it still loses to divide and conquer. As you can see, the trade off between speed and memory is quiet big in this instance, because the running aggregate can become a very high poly object very quickly.

So what is proposed is a trade off. Use more memory for a big speed gain, by prioritizing work with low poly objects as much as possible.

MichaelAtOz, 2015-03-03:
That depends on the Order of the memory increase, there are models which hit the 32 bit memory limits. And on 64 bit that go more than 4GB. Any substantial increase will not be good.

Ghoughpteighbteau, 2015-03-04:
Truthfully? I dunno, I suspect I'd have to profile openSCAD in order to give you a realistic guess. kintel gave me a good guess where to look. https://github.com/openscad/openscad/blob/master/src/cgalutils.cc#L468

I'll do my best to get a development environment set up and make some test projects to profile against.

I'm on arch linux so it's not straightforward for me.

thehans, 2017-04-20:
Union could definitely be sped up a lot by just by changing the order of operations.  For what its worth, I contributed this sort of enhancement to OpenJSCAD project a while ago:  https://github.com/joostn/OpenJsCad/pull/53/files
You can see from my comments and testing, the result was ~260 times faster!!, albeit for a pathological. contrived, union abusing use-case.

I'm fairly certain that a single line change would make this same improvement for CGAL.  
Changing this line:
https://github.com/CGAL/cgal/blob/master/Nef_3/include/CGAL/Nef_nary_union_3.h#L50
to
queue.push_back(tmp);

I don't have time right now to build CGAL and test it all and confirm that everything works, but the idea is that by putting the result of each union at the end of the queue, it works a bit like a binary tree and balances the average complexity of each union between 2 sets.

thehans, 2017-04-20:
Additionally this splitting up the workload could (and should IMO) be tried within OpenSCAD source as well, to take advantage of further multithreading enhancements as being discussed in #1980

thehans, 2018-01-31:
I think my comment about speeding up CGAL by using `push_back` instead of `push_front` was misguided.  After looking some more at the CGAL code, it seems that the following line is effectively ordering the operations in the same way I was suggesting, just doing it in a different and confusing way:
https://github.com/CGAL/cgal/blob/master/Nef_3/include/CGAL/Nef_nary_union_3.h#L57

kintel, 2018-01-31:
That said, scheduling unions in a way that the number of internal vertices are "swallowed" early may be beneficial.

thehans, 2019-11-07:
I took a stab at improving this and just submitted PR #3121.

With regards to performance, I figure I should mention that the test case from the original comment here doesn't seem to make much difference on the current nightly build.  I get 12s to render both the first and the second loop versions, which is great!

So to increase complexity, I changed `$fa=40` to `$fa=10`
```
n=3;
$fa=10;

!for(x=[0:n],y=[0:n],z=[0:n])
translate([x,y,z]*10) sphere(15);

for(x=[0:n]) for(y=[0:n]) for(z=[0:n])
translate([x,y,z]*10) sphere(15);
```
And then I get `1m58s` for the first loop and `1m51s` for the 2nd, so the 2nd is still slightly faster.
Again this was just testing nightly, before my changes.  Not sure what other change might have improved this since the original post, but the difference is almost negligible now compared to the originally reported >200% performance difference.

Of course we can always strive for better performance, so on to discussion of...
**Changes in PR #3121**

@Ghoughpteighbteau wrote:
> what I believe to be ideal behavior isn't possible in userland because a balance between two factors needs to be reached:
> 1. you must union shapes with the least number of vertices first.
> 2. you must union shapes that eliminate vertices as soon as possible.

These changes are currently only addressing point 1.  

I chose to sort each polyhedron by number of face(t)s.  Not sure if this is ideal way to measure complexity in terms of CGAL processing time, but it is easy to change if needed.

The code no longer calls `Nef_nary_union_3`, because it now sorts intermediate results alongside the input geometries.
`Nef_nary_union_3` basically takes one big list and joins objects in groups of powers of 2, which I think should be ideal when all input geometries have same complexity(such as the original example from this issue).  In that case the new code should result in pretty much the same order of operations as before.
Also the new code should be able to avoid copy-construction of all the input polyhedrons that it looks like `Nef_nary_union_3` would need to put them in its list.

Here is a test I made which demonstrates an ideal (and rather contrived 😉) situation for the new code;  a large single union starting with one complex object, followed by many simple ones:
https://gist.github.com/thehans/fbdbc337d9a4fde1d9d3ebacc188e03d
Nightly: 30s render
priority_queue: 14s render

And a more balanced/realistic test running all cgal related tests from our suite against a build from HEAD, then with my changes:  `ctest -C All -R cgal`
I edited away tests that finished in < 1s in both cases and compared times side-by-side in the screenshot below.
Current code from HEAD on the **left**, #3121 on the **right**:
[image]

So most tests were either basically the same time or slightly faster.
Biggest difference was `cgalpngtest_logo_and_text` going from 39.83 sec to 31.71 sec
Overall Total 443.46 sec to 419.25 sec

**Possible further improvements**

One thing that could still be added to my changes would be to merge nested union modules into a single `applyUnion` call, by adding children of a union node to *its* parent when its parent is also a union node.  This should theoretically give the queue a better chance of ordering things optimally.

Another point is that I think this approach could work well to facilitate multithreading of single large unions, as long as the code that accesses the queue is properly locked in a critical section.

And lastly, we could still attempt to address point number 2:  Some way to determine or approximate how many vertices(if any) would be eliminated from a single binary union.

At first thought it seems like a good heuristic would be something like finding the bounding boxes with the most overlap between pairs of polyhedrons, the largest volume of intersection.  
But `CGAL_Nef_polyhedron:BoundingBox` is [not implemented](https://github.com/openscad/openscad/blob/dcc30dda7531ed08bca7bdbffb5c2db3fdaf35c9/src/CGAL_Nef_polyhedron.h#L18) so we'd need that first.  
Also I'm not sure how to combine this with the other ranking by "complexity", and if it would need O(n^2) time to compare all pairs of objects to rate them by volume of intersection.

If we had bounding boxes we could also determine if two given polyhedra intersect at all.  But even with that info, I'm not sure how to combine two Nef_polyhedron_3's  without calling out to `join`(union)

Another idea might be to take the overall distance from the origin for each polyhedron, but what would we use to determine this?  It could be the center of bounding box(again still would have to implement that), or maybe a simple average of polyhedron vertex points which would roughly approximate center of mass?  Either way this approach might not be that great if for example various objects are the same distance from origin but in different octants.  Could we sort in some way that would account for distance *and* direction from origin?  I don't know what that would look like.

Any other ideas for sorting by placement?

kintel, 2024-03-25:
Closing this as we've made a lot of progress with alternatives. Both fast-csg and Manifold give huge performance boosts to unions.

closed 2024-03-25 by kintel