openscad/openscad #1237: Slowdown with large # of dependencies

kintel, 2015-03-05:
From [mailing list](http://forum.openscad.org/2015-02-05-less-responsive-tt11580.html)

When Automatic Reload and Compile is turned on, we regularly scan for changes in dependencies. If the dependency tree is large, this causes a noticable slowdown - typically laggy rendering.
Some ideas:
* If files are included in multiple places, don't check them twice
* Use platform tools for listening for filesystem changes rather than polling manually

nophead, 2015-03-05:
If you currently do check files more than once and recursively then I am sure fixing that alone will be enough to fix the problem. It's akin to the problem with min and max being very slow in recursive functions. The number of files polled will explode exponentially in a project like Mendel90.

Smartlug, 2015-04-04:
Same issue and even blocking when too long... windows state "not responding" in the title, disable window and when re-enabled... reload again...

t-paul, 2015-04-04:
@Smartlug is that a public project somewhere which could be used to check improvements against?

nophead, 2015-04-04:
@t-paul,
   You can use Mendel90. If you open scad/main.scad it pans and scrolls reasonably quickly with auto reload off and becomes almost totally unresponsive with it on. Older versions don't slow down as much but also don't spot all the nested dependencies either. The initial F5 does take about 15 minutes though.

t-paul, 2015-04-04:
@nophead yeah, I know that's the case reported on the mailing list. Having some additional test cases is always good as there could be other cases than the suspected multiple include of the same config file. E.g. with the whole model on a network drive, things get even more tricky.

nophead, 2015-04-04:
It's the fact that the config is included in every file and itself includes a lot of other files. I suspect I could make a skeleton of that with empty files and see the same slowdown since it is only checking mod times, not reading them, so the contents are irrelevant.

I think having the files on a network will be slow anyway because I don't think the stats get cached. Not sure if you can register for changes to non-local files either.

I once wrote a program that needed to monitor dozens of locally networked files. IIRC I just did them one at a time in the idle loop, so it didn't impact the responsiveness of the GUI, but still did them as fast as it could. Not quite the same as this though because it only had to respond to changes when it was actively being interacted with.

nophead, 2015-04-04:
Looking here: https://github.com/openscad/openscad/blob/3c3ac52576297c5bc4b8bacce83f69fa33c730b9/src/ModuleCache.cc#L34 I am wondering if it is worse than I thought. Does it explore the tree of module dependencies and check all the includes of the file that the module resides in? 

In that case it would really explode with a big project.

kintel, 2017-02-02:
Fixed by #1926

closed 2017-02-02 by kintel