openscad/openscad #1246: Crash on a variant of sweep-test.scad

erniejunior, 2015-03-07:
Hi,

I get a total crash of openscad with a variant of the sweep-test.scad from [the list comprehension demos](https://github.com/openscad/list-comprehension-demos).
I am using 2015.03.06.nightly (git ade90b2).
The console output is
```openscad
use <sweep.scad>
use <scad-utils/transformations.scad>
use <scad-utils/shapes.scad>

function func0(x)= 1;
function func1(x) = 30 * sin(180 * x);
function func2(x) = -30 * sin(180 * x);
function func3(x) = (sin(270 * (1 - x) - 90) * sqrt(6 * (1 - x)) + 2);
function func4(x) = 180 * x / 2;
function func5(x) = 2 * 180 * x * x * x;
function func6(x) = 3 - 2.5 * x;

pathstep = 1;
height = 100;

shape_points = square(10);
path_transforms1 = [for (i=[0:pathstep:height]) let(t=i/height) translation([func1(t),func1(t),i]) * rotation([0,0,func4(t)])];
path_transforms2 = [for (i=[0:pathstep:height]) let(t=i/height) translation([func2(t),func2(t),i]) * rotation([0,0,func4(t)])];
path_transforms3 = [for (i=[0:pathstep:height]) let(t=i/height) translation([func1(t),func2(t),i]) * rotation([0,0,func4(t)])];
path_transforms4 = [for (i=[0:pathstep:height]) let(t=i/height) translation([func2(t),func1(t),i]) * rotation([0,0,func4(t)])];
sweep(shape_points, path_transforms1);
sweep(shape_points, path_transforms2);
sweep(shape_points, path_transforms3);
sweep(shape_points, path_transforms4);

path_transforms5 = [for (i=[0:pathstep:height]) let(t=1.5*i/height) translation([0,0,i]) * scaling([func3(t),func3(t),i]) * rotation([0,0,func4(t)])];
#translate([100, 0, 0]) sweep(shape_points, path_transforms5);

path_transforms6 = [for (i=[0:pathstep:height]) let(t=i/height) translation([0,0,i]) * scaling([func6(t),func6(t),i]) * rotation([0,0,func5(t)])];
translate([-100, 0, 0]) sweep(shape_points, path_transforms6);
```
The crash occurred after I added the 1.5 in line 27 (for the path_transforms5).
The accurate output from the console was:
```
Got bus address:  "unix:abstract=/tmp/dbus-R0HucZjJrj,guid=a553d534d292c2a2f915202754f99a44" 
Connected to accessibility bus at:  "unix:abstract=/tmp/dbus-R0HucZjJrj,guid=a553d534d292c2a2f915202754f99a44" 
Registered DEC:  true 
Registered event listener change listener:  true 
openscad-nightly: src/libtess2/Source/priorityq.c:413: pqInit: Assertion `((((TESSvertex *)**(i+1))->s < ((TESSvertex *)**i)->s) || (((TESSvertex *)**(i+1))->s == ((TESSvertex *)**i)->s && ((TESSvertex *)**(i+1))->t <= ((TESSvertex *)**i)->t))' failed.
[1]    24789 abort      openscad-nightly
```

I am happy to answer further questions if needed.

closed 2015-03-08 by kintel

erniejunior, 2015-03-08:
What? I file an issue on github and the next morning I update the Debian package and it is fixed? This is AWESOME!!! Good job @kintel!
The preview (F5) now works nicely. But rendering (F6) is still broken:
The first rendering (and after every Design->Flush Cashes) gives no graphical output and prints
```
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1102 
Geometries in cache: 14
Geometry cache size in bytes: 541632
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 7 seconds
Rendering finished.
```
to the console. Any following compiles give
```
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 14
Geometry cache size in bytes: 541632
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 0 seconds
Rendering finished.
```
Did you check compiling before you closed the issue? Or should I open up a new Issue because it is somewhat different?
Again, I am happy to help out with more information.

kintel, 2015-03-08:
I fixed the crash issue. The remaining issue is a problem with the design:
The design doesn't result in a properly closed volume and thus fails to render - similar to #802, #1184.
I didn't look into the details - this could be a bug in on the of experimental libraries.