openscad/openscad #1254: Fails to locate qscintilla2.prf on Fedora 21

prokoudine, 2015-03-11:
Hi,

Newly released OpenSCAD 2015.03 fails to see /usr/share/qt5/mkspecs/features/qscintilla2.prf and thus fails to build due to the "lack" of QScintilla2 installed.

I have the following packages in my system:

qscintilla-qt5-2.8.4-2.fc21.i686                                                                                   1/2 
qscintilla-qt5-devel-2.8.4-2.fc21.i686 

And the qscintilla2.prf files is exactly where I said it is.

What do I need to tweak to get this fixed?

prokoudine, 2015-03-11:
A quick update: I "fixed" this by running qmake-qt4 instead and hence building with Qt4. Would be nice to not have to do this, though :)

t-paul, 2015-03-11:
The problem here is that a number of distributions have the location of that file either wrong or ship an old QScintilla2 version that does not have the file yet.
We try to workaround the issues (e.g. #938) but have not yet found a generally working solution that does not break other combinations / distributions. Any ideas how to improve the [current way](https://github.com/openscad/openscad/blob/master/scintilla.pri#L13) would be welcome.

nelson18, 2015-04-25:
I have the same problem.  I have openscad-2015.03-1 and qscintilla2.   

The strange thing is that  ./scripts/dependencies.sh  sees that I have qscintilla2 version 2.9.  

depname     minimum     found       OKness      
qt          4.4         5.4.0       OK          
qscintilla2 2.7         2.9         OK          
cgal        3.6         4.1         OK          
gmp         5.0         5.1.1       OK          
mpfr        3.0         3.1.2       OK          
boost       1.35        1.41        OK          
opencsg     1.3.2       1.3.2       OK          
glew        1.5.4       1.7.0       OK          
eigen       3.0         3.2.3       OK          
glib2       2.0         2.36.4      OK          
fontconfig  2.10        2.10.       OK          
freetype2   2.4         2.5.0       OK          
harfbuzz    0.9.19      0.9.19      OK          
bison       2.4         2.4.1       OK          
flex        2.5.35      2.5.35      OK          
make        3           3.81        OK          

But qmake openscad.pro can't find qscintilla2:  

Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
/canopus1/usr/local/src/openscad-2015.03-1/openscad-2015.03-1/scintilla.pri:13: Cannot find feature qscintilla2
Project MESSAGE: Using local copy of qscintilla2.prf instead.

I'm using Qt5:
which qmake
/usr/lib64/qt5/bin/qmake

qmake --version
QMake version 3.0
Using Qt version 5.4.0 in /usr/lib64

Thanks,

Scott

t-paul, 2015-04-25:
@nelson18 The script detects Qt4 and the matching Qt4-QScintilla. If you want to compile with Qt5, that's fine, but you probably have to install the Qt5 development version of QScintilla.

On Debian/Ubuntu that is:
* _Qt4_: libqscintilla2-dev
* _Qt5_: libqt5scintilla2-dev

t-paul, 2015-04-26:
OpenSCAD (or to be specific `qmake-qt5`) can not see the installed qscintilla2.prf in fedora 21 because it's installed in the wrong place.
The included workaround does not work as fedora uses yet another name for the library.

Workaround for Fedora 21: Symlink qscintilla2.prf to the correct place:
```
sudo cd /usr/lib64/qt5/mkspecs/features && ln -s /usr/share/qt5/mkspecs/features/qscintilla2.prf
```

t-paul, 2015-04-26:
@hroncok can you help getting this fixed for the next fedora release?

closed 2015-04-26 by t-paul

hroncok, 2015-04-26:
I was just about to ask if anyone has reported this on bugzilla.

I'll do it, could you give me some link that proves the correct location?
Or the only "proof" is that it works (that'll work as well, minimal
reproducer would be awesome).

hroncok, 2015-04-26:
Before I reporting, just wanted to notify the maintainer of qscintilla

cc @rdieter

hroncok, 2015-04-26:
Found a minimal reproducer.

hroncok, 2015-04-26:
https://bugzilla.redhat.com/show_bug.cgi?id=1215380

t-paul, 2015-04-26:
Thanks @hroncok, I've added another workaround via #1327 which seems to work on 64bit Fedora 21.

Eventually we do need to stop layering workarounds for various distributions though :-).

nelson18, 2015-04-26:
t-paul's  link  command did indeed work.   Now both openscad's  check-dependencies.sh  command and  qmake openscad.pro  are happy.

However, OpenSCAD still doesn't build because my version of boost is newer than expected?   check-dependencies says that boost 1.35 is needed, and correctly detects that I have boost 1.41.  But there is an error:

g++ -c -pipe -fno-strict-aliasing -frounding-math -DEIGEN_DONT_ALIGN -I/usr/local/include/glib-2.0 -I/usr/local/lib64/glib-2.0/include -I/usr/local/include/harfbuzz -I/usr/local/include/freetype2 -I/usr/local/include -I/usr/local/include -I/usr/local/include/freetype2 -I/usr/local/include/libxml2 -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -m64 -mtune=generic -O2 -D_REENTRANT -Wall -W -Wno-unused-local-typedefs -fPIE -DOPENSCAD_VERSION=2015.03-1 -DOPENSCAD_SHORTVERSION=2015.03 -DOPENSCAD_YEAR=.0 -DOPENSCAD_MONTH=.0 -DOPENSCAD_DAY=.0 -DDEBUG -DENABLE_CGAL -DENABLE_OPENCSG -DUSE_SCINTILLA_EDITOR -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_CORE_LIB -I/usr/lib64/qt5/mkspecs/linux-g++ -I. -Isrc -isystem /usr/include/qt5 -Isrc/libtess2/Include -isystem /usr/include/qt5/QtOpenGL -isystem /usr/include/qt5/QtPrintSupport -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtConcurrent -isystem /usr/include/qt5/QtCore -Iobjects -isystem /usr/include/eigen3 -o objects/scintillaeditor.o src/scintillaeditor.cpp
In file included from src/boosty.h:30:0,
                 from src/scintillaeditor.cpp:5:
src/scintillaeditor.cpp: In constructor EditorColorScheme::EditorColorScheme(boost::filesystem::path):
src/scintillaeditor.cpp:79:59: error: boost::filesystem::path has no member named c_str
   PRINTB("Error reading color scheme file '%s': %s", path.c_str() % e.what());

That should be   path.string().c_str()

Ditto in all of the calls to the PRINTDB macro in src/PlatformUtils.cc

Likewise, 

src/scintillaeditor.cpp: In member function void ScintillaEditor::enumerateColorSchemesInPath(ScintillaEditor::colorscheme_set_t&, boost::filesystem::path):
src/scintillaeditor.cpp:373:27: error: boost::filesystem::basic_path<std::basic_string<char>, boost::filesystem::path_traits>::string_type has no member named string
    if (!(path.extension().string() == ".json")) {

should probably be:

    if (!(path.extension() == ".json")) {

Ditto in src/colormap.cc  for both  extension()  and  PRINTB  as shown above.

With these four edits, OpenSCAD appears to compile and run correctly with boost 1.41 (Nov 17 2009).  

The latest boost is 1.58.0 (Apr 17 2015); but I have 1.41 since that is what ships with RHEL 6.6,  2.6.32-504.3.3.el6.x86_64 (my company is very conservative in RHEL updates).

Thanks,

Scott

kintel, 2015-04-26:
@nelson18 Yes, boost 1.41 implements an old version of the filesystem API. We don't regularly test backports against such old versions of boost. It would be helpful if someone could set up some sort of automated build against old Linux distros. Otherwise, we'll just fix things when they're reported.

nelson18, 2015-04-26:
Hum...  but OpenSCAD's  check-dependencies.sh  checker is looking for boost 1.35.  Perhaps the checker needs to be updated to specifically look for something newer?

       boost   1.35   1.41   OK

kintel, 2015-04-26:
We try to be compatible with older versions - we just don't have any CI set up for testing on such old systems.

nelson18, 2015-04-26:
Understood.  

But as-is 2015.03-1 doesn't work with boost 1.41 (and I assume 1.35 either).   Which version of boost are you using?   I'll give that a try.

Thanks,

Scott

kintel, 2015-04-26:
The current Mac binaries use 1.57 - we just moved to 1.58 in master. The Windows binaries use whatever MXE (http://mxe.cc) ships with (currently 1.53), and Linux binaries just use whatever is available on each distro.

rdieter, 2015-04-27:
I can confirm the fedora 21 qt5 qscintilla issue is a (qscintilla) upstream bug wrt handling the qscintilla2.prf location.  Patch to fix it sent to qscintilla list:
http://www.riverbankcomputing.com/pipermail/qscintilla/2015-April/001054.html

link to patch used in fedora packaging:
http://pkgs.fedoraproject.org/cgit/qscintilla.git/plain/QScintilla-gpl-2.9-qt5_mkspecs.patch

hroncok, 2015-04-27:
Thank you @rdieter

commit cc4b633e (2015-05-15): Workaround for Fedora 21 builds (fixes #1254).

Fedora installs qscintilla2.prf in the wrong location and has a different
naming scheme than the one used in Ubuntu so the existing workaround fails.
This adds yet another workaround to detect the different naming for the
Qt5 version of the QScintilla2 library.
commit cd4ced54 (2015-05-15): Merge pull request #1327 from openscad/issue1254-qscintilla-fedora

Workaround for Fedora 21 builds (fixes #1254).
commit 41f6b85c (2015-05-15): Workaround for Fedora 21 builds (fixes #1254).

Fedora installs qscintilla2.prf in the wrong location and has a different
naming scheme than the one used in Ubuntu so the existing workaround fails.
This adds yet another workaround to detect the different naming for the
Qt5 version of the QScintilla2 library.