openscad/openscad #1256: Can't pass child modules to children()

davidson16807, 2015-03-12:
I've been trying to implement a new feature for my [library](https://github.com/davidson16807/relativity.scad). I'd like to create a module called `anchor()`. Given a child module that invokes the `children()` operator, `anchor()` translates the child module so that its origin is located at a location within the child module that's specified by a given string parameter.

As an example:

    anchor("foo")
    box(50)
    align(top)
    ball(50)
    align(top)
    rod(50, $class="foo")
    children();

The object above will be rendered with its origin located at the object of class "foo".

The implementation I worked out requires that `anchor()` invoke its child module twice. The first invocation calculates the absolute position at location "foo" without being rendered. The second invocation translates and renders the child module with an offset of -2 times the position found from the first invocation. It looks something like this:

module anchor(class){
    attach(class)
    children()
    translate(-2*$position)
    children();
} 

`$position` is a special variable representing absolute position as calculated by some of the objects in my library. `attach(class)` just allows only the second instance of children() to be rendered.

The problem, I discovered, is that OpenSCAD does not appear to support passing child modules to the children() operator. I've made sure of this by testing the most basic example:

module foo(){
    children()
    children();
} 

Its possible this has been discussed before, but I don't see it when I search existing issues. Is this intended behavior? Is there any strong reason this can't/shouldn't be implemented?

nophead, 2015-03-12:
Not sure I fully understand what you are trying to achieve. Do you want to pass a child to itself and have it position itself and the also render itself? E.g.
```
module kid()
   if($position)
       translate($position) {
           $position = false;
           children();
       }
   else
       cube(10);
   

module foo() {
   $position = [0, 10, 20];
   children() 
        children();
}

foo() kid() kid();
```
This does what I expect. I.e. a very confusing way to do translate([0, 10, 20]) cube(10);

davidson16807, 2015-03-12:
The kid() example works because kid() is called twice on the bottom line.

Rather, I'd expect the following code to render two cubes:

    module kid(){
       cube(10);
       translate([10, 10, 10])
           children();
    }
    
    module foo() {
       children() 
            children();
    }
    
    foo() kid();

This only renders a single cube in 2015.03. 

The goal of my example function is to allow the alignment of complex geometries. You'd define two modules with any number of nested translates, submodules, or primitives, then specify a number of attachment points within them. Each attachment point has a string based identifier. You'd invoke the anchor parameter, passing it an id for the point of attachment, then it would place that attachment point at the origin. This would allow you to take two arbitrarily complex modules and stick them together without having to design the individual modules around that attachment point.

kintel, 2015-03-12:
@davidson16807 One challenge is that children() return a union of children (or rather a group of children which results in an implicit union). This union already receives the original children as parameter, and there's not way another set of children can be passed as parameters.

I still don't 100% understand what you're trying to achieve. e.g. what would happen if multiple children were passed?

nophead, 2015-03-12:
Hmm, very confusing.
```
foo() kid() sphere(5);
```
Shows a cube and a translated sphere. So kid already potentially has a child before it is passed to foo().

davidson16807, 2015-03-12:
@kintel The problem with implicit union is understandable. 

If multiple children were passed, It makes sense to me that each member of children() would be passed the entire set of children().

@nophead See, from my uninitiated point of view, I'd expect that to be equivalent to `kid() sphere(5)`. The way I think, the children of foo() are analogous to a lambda expression that you pass to a higher order function. It would be much like `foo(lambda children: kid(sphere())`. You would need to call foo(lambda children: kid(sphere(children()))) in order to render 2 spheres and 2 cubes. In OpenSCAD that translates to `kid() sphere(5) children()`. However, this contradicts the behavior I intuitively expect from the following:

    module bar(){
        foo() kid() sphere(5) children();
    }

Which could be interpreted as either `bar = lambda children: foo(kid(sphere(children)))` or `bar = lambda children: foo(lambda children2: kid(sphere(children2)))`. So yeah, that would be undefined behavior. I suppose the lack of clarity stems from there not being a way to assign names for a specific instance of children, as you would with a lambda in the above examples.

nophead, 2015-03-12:
I am afraid you have completely lost me but sphere(5) children() will ignore children() because the sphere module ignores its children.

The point I was making is if kid() picks up its children from where it is instiated it cant be passed more children in foo when it is accessed through children(). The meaning of the construct could be one or the other but not both.

davidson16807, 2015-03-12:
Yeah, sorry. I went too far with the sphere example. You could consider another function in place of `sphere` that does accept children, e.g. 

    module bar(){
        foo() kid() kid2() children();
    }
    module kid2(){
        sphere(5);
        translate([10,10,10])
            children();
    }

I'll stop mention the thing about lambdas. What I was saying was it could work if you treat the descendants of `foo` as analogous to a module, so `foo() kid() kid2()` would be equivalent to `foo() descendants()` where `descendants` is:

    module descendants(){
        kid() kid2();
    }

In other words, `descendants()` would render only once since its doesn't invoke its own `children()`. You could make it so that you'd call `foo() kid() kid2() children()`, but that doesn't work because it contradicts what you expect from implementing `bar()`, above. You'd need to make a decision on what that `children()` refers to, and to keep the existing behavior it would have to represent the children passed to `bar`. 

In other words, you could still make it work, calling `children() children()`, but `foo() kid() kid2()` would have to be like calling `kid() kid2()`

closed 2026-01-29 by github-actions[bot]