openscad/openscad #1259: Support patch level releases of major versions

kintel, 2015-03-13:
As OpenSCAD tends to get a major release relatively seldomly (1-2 times a year), it would be beneficial to the community if we could release patch level updates with fixes for critical issues, without having to wait a full release cycle. This could also be particularly interesting for Linux distribution packaging.

Things needing resolution && TODO:
* [ ] Version numbering scheme (e.g. 2015.03-1 ?)
* [ ] Auto-update behavior (OS X): We use a date-tagged release. Make sure this doesn't clash with snapshot updates
* [ ] How should version number scheme co-exist with Linux packages, which tend to have their own patch level release number scheme?
* [ ] Define branching & tagging scheme on the git repos
* [ ] Update release checklists
* [ ] Update build system and scripts to handle new release numbering scheme

hroncok, 2015-03-13:
I would appreciate if the version is just 2015.03.x without any dash. But
that would, of course, conflict with devel versions based on date.

If you choose to use 2015.03-x I can append the patch level release number
like this. 2015.03-x.y.

Also, those releases would be great,so +1000 from me.

For git branch, why just not name the branch 2015.03.x or 2015.03-x
depending on the version scheme? With actual letter x in there, here I
don't mean it as a placeholder.

kintel, 2015-03-13:
I feel that 2015.03.x would be too confusing as the x would represent a patch-level incremental number rather than a date.

hroncok, 2015-03-13:
I agree.

closed 2015-03-21 by kintel