openscad/openscad #1261: Compile fails on previously working scad file after upgrade to 2015.03

mak3r, 2015-03-14:
I've upgraded to 2015.03 on OS X today and one of my recent scad files is no longer compiling. Unfortunately, I don't recall what version I was using before. It was the most up to date prior version. 

ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: cet->get_index() == ce->twin()->get_index() File: /Users/kintel/code/OpenSCAD/openscad-2015.03/../libraries/install/include/CGAL/Nef_3/SNC_external_structure.h Line: 1169 

The complete code 
```
hulled_resize();

//goal size
//#cube([73,14,143],center=true);

module forced_scale() {
	//goal size
	//#cube([73,14,143],center=true);
	difference() {
		scale([1.09,2.1,1.04])
		import("iphone-6_clean.stl");

		translate([0,-9.7,0])
		cube([73,5,143],center=true);
	}
}

module hulled_resize() {
	resize([73,12,143])
	hull() {
		difference() {
			union() {
	//			color("blue")
				mirror([1,0,0])
				import("iphone-6_clean.stl");
	//			color("green")
				mirror([1,0,0])
				rotate(180,[0,1,0])
				import("iphone-6_clean.stl");
	//			color("red")
				rotate(180,[0,1,0])
				import("iphone-6_clean.stl");
				import("iphone-6_clean.stl");
			}
		
			translate([0,-5.5,0])
			cube([73,5,143],center=true);

			translate([0,5.5,0])
			cube([73,5,143],center=true);
		}
	}
}
```

I'm happy to share the stl if needed for debugging. The original was 'fixed' by netfabb but still had issues so I fixed it manually in meshlab. Perhaps it still has issues that I did not uncover.

p.s. I love the new interface. Nice work!

kintel, 2015-03-14:
@mak3r Please share the STL file - this issue is directly related to STL import.

mak3r, 2015-03-14:
@kintel How can I submit the stl to you?

kintel, 2015-03-14:
How big is it? email, dropbox, gist?

mak3r, 2015-03-15:
Too big for gist. The stl and compressed stl are here https://www.dropbox.com/sh/nz2om1w4jc15fhl/AAAWFC5EGzde_EdU0fB1ba35a?dl=0

kintel, 2015-03-16:
As far as I can tell, the imported design isn't a solid volume - it has a large cutout in the front. This causes any operation on it to likely fail, and you end up with an undefined result. It looks like what you want in this case is to immediately take the hull() of the imported object, to make sure that it can be used in subsequent boolean operations.

While this technically displays changed behavior between 2014.03 and 2015.03, maintaining "bug compatibility" between versions is not something we're capable of undertaking.

closed 2015-03-16 by kintel