openscad/openscad #1268: [2015.03] Close without Saving etc buttons are untranslated

hroncok, 2015-03-17:
Just came across this:

[image]

And I believe I've translated the entire .po file, so I guess those strings (button labels) are just not gettexted.

kintel, 2015-03-17:
Those probably come from Qt. Are other Qt-based apps translated correctly?

hroncok, 2015-03-17:
Tried with Qt4 Designer:

[image]

kintel, 2015-03-17:
ok, so somehow we're not picking up Qt translations. Is OpenSCAD also build using Qt4?
@teepee Does this ring any bells?

t-paul, 2015-03-17:
I'd have expected the Qt translation files to be used automatically (same for QScintilla). But maybe we have to do some additional configuration or something? All that is not going though gettext but uses the Qt's own translation logic. For Windows/OSX we might even need to ship the translation files...

I just checked, and I'm getting the same result on Debian.

hroncok, 2015-03-18:
@kintel Yes, OpenSCAD is also built with Qt4

closed 2026-01-31 by github-actions[bot]